f4pga/f4pga-arch-defs #578: Improve xc7 fasm2verilog output

litghost, 2019-04-11:
With https://github.com/SymbiFlow/symbiflow-arch-defs/pull/577, fasm2v is good enough to be useful, however it could be more useful when debugging correctness issues with some more features.

Ideas to improve output, along with difficulty:
- [ ] (easy) Combine carry chain inputs and outputs into wire bus for easier viewing in during simulation.  This is a pretty straight forward verilog parsing task, basically join the various CARRY4's that are connected via the chain and emit 4 wire buses, S, DI, CO and O for the entire chain length.
- [ ] (easy) Merge simple nets, leaving the intermediate wires, but avoiding the chain.  So
```
assign CLBLM_R_X11Y110_SLICE_X14Y110_CMUX = CLBLM_R_X11Y110_SLICE_X14Y110_C5Q;
assign CLBLL_R_X17Y120_SLICE_X26Y120_SR = CLBLM_R_X11Y110_SLICE_X14Y110_CMUX;

Becomes:

assign CLBLM_R_X11Y110_SLICE_X14Y110_CMUX = CLBLM_R_X11Y110_SLICE_X14Y110_C5Q;
assign CLBLL_R_X17Y120_SLICE_X26Y120_SR = CLBLM_R_X11Y110_SLICE_X14Y110_C5Q;

```
- [ ] (more difficult) Name nets with same name that VPR assigned them.  These names are found in the route output from VPR.  This involves consuming the VPR route output which looks like:

```
Net 0 (scalable_proc.uart.reg_dat_di[31])                                      
                                                                               
Node: 69480 SOURCE (29,26)  Class: 0  Switch: 0 
```

And then matching the VPR rr graph source node OPIN back into a 7-series source, and then renaming the net.

mkurc-ant, 2019-04-17:
@litghost Regarding "Add a comment above LUT6_2's with the equation form of the LUT from input to output. Ideally they will be reduced equations.":

You mean to write own implementation of an algorithm which converts a truth table with up to 6 inputs to a boolean equation ? Or use an already available solver for this task ? Pure python or an external library/tool ?

I would gladly go with this task, but the only solution I see right now is to use Karnaugh tables related stuff. Or do you have something else on your mind ?

litghost, 2019-04-17:
> @litghost Regarding "Add a comment above LUT6_2's with the equation form of the LUT from input to output. Ideally they will be reduced equations.":
>
> You mean to write own implementation of an [...]

Yosys can perform this composition, and I believe there are python libraries too.
 
> I would gladly go with this task, but the only solution I see right now is to use Karnaugh tables related stuff. Or do you have something else on your mind ?

That is what I was thinking, about it doesn't have to be all implemented by us.

mkurc-ant, 2019-04-19:
@litghost 

> Yosys can perform this composition
I believe that Yosys can only do the opposite: convert boolean equation to a truth table. Not vice-versa.

I found some implementation of Karnaugh map solvers in python but they are limited to 4 inputs only. We are aiming at LUTs with up to 6 inputs (as they are in 7-series) right? A Karnaugh solver is very straightforward so I can write own implementation for 6-input LUTs.

There is also the Quine-McCluskey algorithm which will always find the optimal solution but it is too much time consuming. AFAIK works for any number of inputs. Implementations are available.

I found also a heuristic algorithm called "Espresso". There is source code available: https://ptolemy.berkeley.edu/projects/embedded/pubs/downloads/espresso/index.htm

I'll keep looking for other solutions. I guess I'll go with this task.

litghost, 2019-05-17:
In order to better compare timing information output from VPR and Vivado, it is good to annotate the BEL names with the VPR names.  I'll look into improving that next.  This should enable side-by-side timing path comparisons.

mithro, 2019-05-17:
@mkurc-ant BTW I found the following - [Generate logic expressions for all 64K boolean functions of four variables A, B, C, D](https://github.com/shirriff/xc2064/blob/master/formula4.py)

> # Generate logic expressions for all 64K boolean functions of four variables A, B, C, D
> # The idea is to start with simple expressions, then take all combinations of these to
> # generate new expressions, then all combinations of the new expressions, etc. until
> # everything has been generated.
> #
> # Formulas are built up as trees, e.g. ['+', 'A', 'B'] is "A or B"