f4pga/f4pga-arch-defs #595: v2x features to replace XML generation

litghost, 2019-04-16:
As part of the BEL timing improvements, several new V2X features are required:
- [ ] Add support for arrays of modules (e.g. num_pb's > 1)
- [ ] Add support for FASM prefix on arrays of modules
- [ ] Add support for FASM LUT annotations
- [ ] Add support for FASM mux annotations on ``<direct>`` and ``<mux>`` within ``<interconnect>``
- [ ] Add support for FASM features on ``<mode>``
- [ ] Add support for FASM parameters on primative ``<pb_type>``'s

litghost, 2019-04-16:
@elmsfu @kgugala FYI

elmsfu, 2019-04-16:
@litghost @kgugala Any objections to using attribute annotations for example`(* SYMBIFLOW_FASM_PREFIX = feature_name *)`

I think using we should consider adding a prefix to the timing information as well, so it is obvious that it is a symbiflow construct. maybe instead of symbiflow `V2X`, `verilog_arch_def`, or `vtr`. thoughts?

litghost, 2019-04-16:
> @litghost @kgugala Any objections to using attribute annotations for example`(* SYMBIFLOW_FASM_PREFIX = feature_name *)`
>
> I think using we should consider adding a prefix to the timing information [...]

I think thats a solid idea.  What about ``FASM.PREFIX`` over ``FASM_PREFIX``.  ``_`` is roughly a space and ``.`` is rough "hierarch" or "namespace".

mithro, 2019-04-16:
I would drop the `SYMBIFLOW_` bit?

elmsfu, 2019-04-16:
@mithro I was thinking that it would be good to use a prefix for all symbiflow or arch-def specific attributes so it was obvious they are specific to generating arch-defs and not more broadly supported or used by some other tool. Do you think it's too verbose?

litghost, 2019-04-16:
> Do you think it's too verbose?

Too verbose.  ``FASM.`` or ``V2X.`` is probably sufficient.

elmsfu, 2019-04-16:
What about for the timing info. Useful to have a prefix? or is there a standard practice?

kgugala, 2019-04-16:
write now those are added with:

```
(* DELAY_MATRIX_<input_port> = ... *)
(* DELAY_CONST <input_port> = ... *)
```
I think those are quite explanatory

kgugala, 2019-04-24:
Initial code for handling multiple instances of the same cell is a part of the #639

mkurc-ant, 2019-05-06:
I am taking adding FASM prefix support to the v2x

mkurc-ant, 2019-05-06:
Preliminary implementation of FASM feature support for V2X: https://github.com/SymbiFlow/symbiflow-arch-defs/pull/685

mkurc-ant, 2019-05-08:
https://github.com/SymbiFlow/symbiflow-arch-defs/issues/697