f4pga/f4pga-arch-defs #599: v2x solve issues to have working tests

acomodi, 2019-04-16:
Currently v2x tests (utils/vlog/tests) are being disabled as the only way to make all of them pass was to have some workarounds in the `vlog_to_pbtype.py` scripts.

The workarounds should be avoided with cleaner solutions and then we can re-enstablish the v2x tests

mithro, 2019-04-19:
@acomodi Are you working on this?

mithro, 2019-04-20:
First issue is;
```
[  0%] Generating dsp_combinational_pb_type_diff.diff
--- /usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs/build/utils/vlog/tests/dsp_combinational/dsp_combinational.pb_type.golden.xml 2019-04-19 19:50:52.160854352 -0700
+++ /usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs/build/utils/vlog/tests/dsp_combinational/dsp_combinational.pb_type.actual.xml 2019-04-19 19:50:52.068855422 -0700
@@ -1,5 +1,5 @@
 <?xml version="1.0"?>
-<pb_type xmlns:xi="http://www.w3.org/2001/XInclude" blif_model=".subckt dsp_combinational" name="dsp_combinational" num_pb="1">
+<pb_type xmlns:xi="http://www.w3.org/2001/XInclude" blif_model=".subckt dsp_combinational" name="BLK_IG-dsp_combinational" num_pb="1">
   <input name="a" num_pins="32"/>
   <input name="b" num_pins="32"/>
   <input name="m" num_pins="1"/>
utils/vlog/tests/dsp_combinational/CMakeFiles/dsp_combinational_pb_type_diff.dir/build.make:63: recipe for target 'utils/vlog/tests/dsp_combinational/dsp_combinational_pb_type_diff.diff' failed   
```

mithro, 2019-04-20:
[Remove `BEL_FF` (etc) naming #561](https://github.com/SymbiFlow/symbiflow-arch-defs/issues/561)

mithro, 2019-04-20:
@acomodi The first issue listed above should be solved by the pull request at https://github.com/SymbiFlow/symbiflow-arch-defs/pull/613

acomodi, 2019-04-20:
@mithro Currently I am not actively dealing with re-establishing the v2x tests as I am more focused on the tile equivalence issue. As far as I know, though, it should not require too much effort to get tests working again, so I could deal with this as well.

mithro, 2019-04-23:
#561 has been merged now, I'm looking at the tests.

mithro, 2019-04-23:
Current set of tests;
 - [ ] `fig41-full-adder`
 - [ ] `fig42-dff`
 - [ ] `dsp_combinational`
 - [ ] `dsp_in_registered`
 - [ ] `dsp_inout_registered`
 - [ ] `dsp_inout_registered_dualclk`
 - [ ] `dsp_modes`
 - [ ] `dsp_out_registered`
 - [ ] `dsp_partial_registered`
 - [ ] `simple_pll`

mithro, 2019-04-23:
These tests will need to be added;
 - [ ] #630 - Add v2x test for Figure 43 - Mixed sequential/combinational single port ram
 - [ ] #631 - Add v2x test for Figure 44 - Sequential single port ram
 - [ ] #632 - Add v2x test for Figure 45 - Sequential single port ram with a combinational input
 - [ ] #629 - Add v2x test for Figure 46 - Multiclock Dual Port RAM

mithro, 2019-04-23:
I created a [SymbiFlow Parts Encyclopedia](https://docs.google.com/document/d/1ngck6LGVTXm6HDrVqWeEGdEELrKPWXVaeejkXurgfl4/edit#) document which is suppose to cover how various verilog and xml stuff is related.

I've only really had a chance to start filling out routing muxes so far...

mithro, 2019-04-23:
For the remaining tests the issue seems to be related to when the inputs are registered.

The following verilog;
```verilog
`include "../dsp_combinational/dsp_combinational.sim.v"

module dsp_in_registered (clk, a, b, m, out);
        localparam DATA_WIDTH = 64;

        input wire clk;
        input wire [DATA_WIDTH/2-1:0] a;
        input wire [DATA_WIDTH/2-1:0] b;
        input wire m;
        output wire [DATA_WIDTH-1:0] out;

        reg [DATA_WIDTH/2-1:0] q_a;
        reg [DATA_WIDTH/2-1:0] q_b;
        reg q_m;
        always @(posedge clk) begin
                q_a <= a;
                q_b <= b;
                q_m <= m;
        end

        dsp_combinational comb (.a(q_a), .b(q_b), .m(q_m), .out(out));
endmodule
```

Ends up with the following error;
```
[  0%] Generating dsp_inout_registered.pb_type.xml 
is_blackbox False has_modes? False                   
Traceback (most recent call last):                  
  File "/usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs/utils/vlog/vlog_to_pbtype.py", line 461, in <module>
    sys.exit(main(args))                            
  File "/usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs/utils/vlog/vlog_to_pbtype.py", line 450, in main                                           
    pb_type_xml = make_pb_type(yj, tmod)                                           
  File "/usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs/utils/vlog/vlog_to_pbtype.py", line 391, in make_pb_type
    make_pb_content(yj, mod, pb_type_xml, mod_pname)                                                   
  File "/usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs/utils/vlog/vlog_to_pbtype.py", line 221, in make_pb_content
    .format(pb_name, pin, mod)                                                                                                                                          
AssertionError: ERROR: pin dsp_combinational.a[0] has no driver, interconnect will be missing
YosysModule(dsp_inout_registered,                                                                                                                                
{'attributes': {'dynports': 1,                           
```

This could be because [`proc` step](http://www.clifford.at/yosys/cmd_proc.html) has not be run in yosys and thus the always block has not been converted to flipflop / latches / etc?

mithro, 2019-04-23:
FYI - This is mentioned in the [utils/vlog/yosys/json.py](utils/vlog/yosys/json.py) file below;
https://github.com/SymbiFlow/symbiflow-arch-defs/blob/48be48e95dd972047327698d8721a46639504f4e/utils/vlog/yosys/json.py#L1-L14

mithro, 2019-04-23:
Looks like the command trying to be run is;
```
cd utils/vlog && /usr/bin/cmake -E env YOSYS=build/env/conda/bin/yosys build/env/conda/bin/python3 utils/vlog/vlog_to_pbtype.py --top=dsp_in_registered -o build/utils/vlog/tests/dsp_in_registered/dsp_in_registered.pb_type.xml build/utils/vlog/tests/dsp_in_registered/dsp_in_registered.sim.v 
```

mithro, 2019-04-24:
Adding a `V=1` will get you the Yosys commands being run. They are;
```
['/usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs/build/env/conda/bin/yosys', '-q', '-p', 'read_verilog -DPB_TYPE  /usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs/build/utils/vlog/tests/dsp_in_registered/dsp_in_registered.sim.v; prep ; write_json ']
['/usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs/build/env/conda/bin/yosys', '-q', '-p', 'read_verilog -DPB_TYPE -DPB_TYPE  /usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs/build/utils/vlog/tests/dsp_in_registered/dsp_in_registered.sim.v; prep ; write_json ']
is_blackbox False has_modes? False                 
['/usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs/build/env/conda/bin/yosys', '-q', '-p', 'read_verilog -DPB_TYPE -DPB_TYPE  /usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs/build/utils/vlog/tests/dsp_in_registered/dsp_in_registered.sim.v; proc; cd dsp_in_registered; select -write /tmp/tmpx16wtllc c:* %x:+[CLK]:+[clk]:+[clock]:+[CLOCK] a:CLOCK=1 %u c:* %d x:* %i']
```

mithro, 2019-04-24:
Looks like proc / prep might already been called, so must be something else?

mithro, 2019-04-24:
When I run yosys manually, I don't get anything particularly concerning...
```
build/env/conda/bin/yosys -q -p 'read_verilog -DPB_TYPE build/utils/vlog/tests/dsp_in_registered/dsp_in_registered.sim.v; prep ; write_json'
```

mithro, 2019-04-24:
Adding `cd dsp_in_registered; show` shows a perfectly reasonable diagram of the following;
[image]

kgugala, 2019-04-24:
@mithro the `dsp_in_registered` fails because we're missing dff's there. To be more precise https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/utils/vlog/yosys/json.py#L248 returns only drivers which are either inputs to the module, or output of another cell.

Since the dffs in https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/utils/vlog/tests/dsp_in_registered/dsp_in_registered.sim.v#L15 are implicit, the script fails with missing net driver.

Adding explicit dff instantiation seems to be solving this issue, but we get hit by multilple pb_type bug.

kgugala, 2019-04-24:
#639 solves the issue with `dsp_in_registered`, however there is still a problem with pb_num in the pb_model.xmls

@mithro please take a look on #639 if it is OK, I'll fix all the rest of dsp registered tests (they all fail with the same issue)

elmsfu, 2019-04-24:
Is [this](https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/utils/vlog/yosys/json.py#L80) doing the correct thing for vectors? I think we are just picking the first one for each?

elmsfu, 2019-04-24:
If I understand we should be able to infer the dependency, but need to add logic to traverse the netlist graph to determine it for cases that any module doesn't have a pb_type.

mithro, 2019-04-25:
Two comments;
 * The version you have after your fix **should** work and be supported.
 * The version before you fix should **also** work.

As I mentioned in #639;

> The verilog `always` shown below (which is called a "process")
> ```verilog
> always @(posedge clk) begin
>    ....
> end
> ```
> **should** cause Yosys to generate `$dff` objects once the `proc` pass is run (the `proc_dff` pass is the exact one).
> 
> This makes me think that we are missing `proc` somewhere as I described above....

mithro, 2019-04-25:
@kgugala - Why are the auto-generated `$dff` cells being ignored? They should be like any other cell?

mithro, 2019-04-25:
So the issue might be these lines;
https://github.com/SymbiFlow/symbiflow-arch-defs/blob/2a41dcc11eee49c3315d811e7060409c4a9fac56/utils/vlog/yosys/json.py#L229-L236

The code seems to be striping out cells which start with `$` like the `$dff` cells generated by the `proc` pass...

mithro, 2019-04-26:
I've added bug https://github.com/SymbiFlow/symbiflow-arch-defs/issues/599 to track the implicit `dff` stuff. Lets land https://github.com/SymbiFlow/symbiflow-arch-defs/pull/639 with the explicit dff implimentation.