f4pga/f4pga-arch-defs #612: ModuleNotFoundError: No module named 'intervaltree'

mithro, 2019-04-20:
Ran `$ make test_python_utils` and got the follow error;
```
================================================================================ ERRORS ================================================================================
_____________________________________________________________ ERROR collecting lib/connection_database.py ______________________________________________________________
lib/connection_database.py:3: in <module>
    from lib.rr_graph import graph2
lib/rr_graph/graph2.py:7: in <module>
    from lib.rr_graph import channel2
lib/rr_graph/channel2.py:8: in <module>
    from intervaltree import IntervalTree, Interval
E   ModuleNotFoundError: No module named 'intervaltree'
______________________________________________________________ ERROR collecting lib/rr_graph/channel2.py _______________________________________________________________
lib/rr_graph/channel2.py:8: in <module>
    from intervaltree import IntervalTree, Interval
E   ModuleNotFoundError: No module named 'intervaltree'
_______________________________________________________________ ERROR collecting lib/rr_graph/graph2.py ________________________________________________________________
lib/rr_graph/graph2.py:7: in <module>
    from lib.rr_graph import channel2
lib/rr_graph/channel2.py:8: in <module>
    from intervaltree import IntervalTree, Interval
E   ModuleNotFoundError: No module named 'intervaltree'
_____________________________________________________________ ERROR collecting lib/rr_graph_xml/graph2.py ______________________________________________________________
lib/rr_graph_xml/graph2.py:2: in <module>
    from lib.rr_graph import graph2
lib/rr_graph/graph2.py:7: in <module>
    from lib.rr_graph import channel2
lib/rr_graph/channel2.py:8: in <module>
    from intervaltree import IntervalTree, Interval
E   ModuleNotFoundError: No module named 'intervaltree'
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! Interrupted: 4 errors during collection !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
======================================================================= 4 error in 0.38 seconds ========================================================================
utils/CMakeFiles/test_python_utils.dir/build.make:57: recipe for target 'utils/CMakeFiles/test_python_utils' failed
make[4]: *** [utils/CMakeFiles/test_python_utils] Error 2
make[4]: Leaving directory '/usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs/build'
CMakeFiles/Makefile2:4647: recipe for target 'utils/CMakeFiles/test_python_utils.dir/all' failed
make[3]: *** [utils/CMakeFiles/test_python_utils.dir/all] Error 2
make[3]: Leaving directory '/usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs/build'
CMakeFiles/Makefile2:4654: recipe for target 'utils/CMakeFiles/test_python_utils.dir/rule' failed
make[2]: *** [utils/CMakeFiles/test_python_utils.dir/rule] Error 2
make[2]: Leaving directory '/usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs/build'
Makefile:1540: recipe for target 'test_python_utils' failed
make[1]: *** [test_python_utils] Error 2
make[1]: Leaving directory '/usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs/build'
Makefile:20: recipe for target 'test_python_utils' failed
make: *** [test_python_utils] Error 2
```

mithro, 2019-04-20:
If you run a `make test_python` it works fine and then `make test_python_utils` works....

mithro, 2019-04-20:
I think this is a dependency problem?

mithro, 2019-04-20:
Any ideas @elmsfu ?

elmsfu, 2019-04-22:
I think we will continue to have such issues until we tackle #463

PR #615 (2019-04-22): utils: add intervaltree to test dependencies
Fixes #612
commit 726ddd51 (2019-04-22): utils: add intervaltree to test dependencies

Fixes #612
closed 2019-04-22 by kgugala