f4pga/f4pga-arch-defs #617: {S,D}PRAM128 name is misleading and not in the CLB User Guide

HackerFoo, 2019-04-22:
For example, a RAM128X1S is built from 2 6 input LUTS (2 DPRAM64s or a SPRAM64 and a DPRAM64) in the CLB User Guide, but [cells_map.v](https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/xc7/techmap/cells_map.v#L263-L311) calls them {S,D}PRAM128s.

mithro, 2019-04-23:
I agree that these names are misleading. I'm thinking the reason they are name `DPRAM128` is because they are being used to make the `RAM128X`? If that is the case then we could use something like `DPRAM64_for_RAM128X1S`?

HackerFoo, 2019-04-23:
It also seems like there is a lot of duplication between the RAM64 variants.

litghost, 2019-04-27:
> For example, a RAM128X1S is built from 2 6 input LUTS (2 DPRAM64s or a SPRAM64 and a DPRAM64) in the CLB User Guide, but [cells_map.v](https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/xc7/techmap/cells_map.v#L263-L311) calls them {S,D}PRAM128s.

The 128s have separate subckt names because of the WA7USED configuration bit.  In the case of RAM128X1S, two RAM128X1S instances can be packed into one SLICEM, however is not legal to pack RAM128X1S and say RAM64X1D, because the WA7USED configuration is SLICEM wide.  RAM128X1D and RAM256X1S do not have this problem because they consume the entire SLICEM, so no ambiguous packing can occur.

@mithro suggestion of DPRAM64_for_RAM128X1S is fairly accurate.

mithro, 2019-04-28:
@HackerFoo can you look at fixing this up?

closed 2019-04-30 by HackerFoo