f4pga/f4pga-arch-defs #619: Documentation implies multiple LUT modes, but there is only one.

HackerFoo, 2019-04-23:
https://github.com/SymbiFlow/symbiflow-arch-defs/blob/bcdfe977bf2c7761174ecea91b25a71d10cb056b/xc7/primitives/common_slice/Nlut/ntemplate.Nlut.pb_type.xml#L3-L7

mithro, 2019-04-23:
It appears that the LUT6 is being modelled as a 2 * 5 input LUTs an extra MUX to combined them.

mithro, 2019-04-23:
This can be confirmed by look at the techmap;
https://github.com/SymbiFlow/symbiflow-arch-defs/blob/619364d5ddfc608a9e38f9290e4cad045db92596/xc7/techmap/cells_map.v#L240-L258

mithro, 2019-04-25:
@HackerFoo - Here is some older stuff -> https://github.com/SymbiFlow/symbiflow-arch-defs/pull/647

Need to talk to @litghost about why it was disabled....

litghost, 2019-04-27:
The true underlying hardware is not configurable, but actually is simple two LUT5's, with the O5 output as the "lower" LUT5 and the upper the O6 output the output for the two LUT6's combined with a mux similar to the MUXF7.  Adding a mode between the LUT5x2 and LUT6 does not increase VPR's packing ability, as any LUT6 would be packed as a LUT5x2 + MUXF6 (which has a pack pattern for this purpose).  There is a mode on the MUXF6 specifically to enable VPR to use the lower and/or upper LUT's as route through LUTs.

In addition, the LUT6_2 mode of the SLICEL is probably better expressed as LUT5x2 + MUXF6, rather than as a 4 mode.

During testing, I found that having 3 modes here confused VPR more than it helped anyways.  Because LUT5x2 + MUXF6 is equivalent to the LUT6_2, and is in fact a more accurate description, I recommend we do not add the LUT6 mode back.

There is one ability that VPR lacks in the current configuration, and that is using the high LUT pin as a route through wire.  I believe the high LUT pin is the fastest pin, so there is reason to explore how to expose to VPR routing through this high pin.