f4pga/f4pga-arch-defs #621: Why is there a bram_l and bram_r which are identical?

mithro, 2019-04-23:
* [bram_l](https://github.com/SymbiFlow/symbiflow-arch-defs/tree/master/xc7/primitives/bram_l)
 * [bram_r](https://github.com/SymbiFlow/symbiflow-arch-defs/tree/master/xc7/primitives/bram_r)

Seem to map from the same set of top level ports to the [bram primitive](https://github.com/SymbiFlow/symbiflow-arch-defs/tree/master/xc7/primitives/bram_r).

Just seems an extra level for no reason?

mithro, 2019-04-23:
```xml
  <direct name="RAMB18_X0Y1_ADDRARDADDR"      input="BLK_IG-BRAM_L.RAMB18_X0Y1_ADDRARDADDR"         output="BLK_IG-BRAM.RAMB18_Y1_ADDRARDADDR" />
```
verse
```xml
  <direct name="RAMB18_X0Y1_ADDRARDADDR"      input="BLK_IG-BRAM_R.RAMB18_X0Y1_ADDRARDADDR"         output="BLK_IG-BRAM.RAMB18_Y1_ADDRARDADDR" />
```

litghost, 2019-04-29:
The routing import currently uses tile wires, which tend to vary between _L and _R tile variants, at least in the CLBLL_L/R examples.  Because the BRAM's have the BRAM_INT_INTERFACE tiles, the BRAM tile wires for the L/R variants may in fact be identical, I haven't looked.