f4pga/f4pga-arch-defs #622: Write documentation about why the stub is needed to create dual port 128 LUT RAM

mithro, 2019-04-23:
@HackerFoo Can you add details here?

I think we determined it was because you need need two pairs of LUTs which must all be placed in the same SLICE, however there is nothing about the LUTs to tell VPR they should be in the same tile.

The stub takes the `SPO` output from one LUT pair and `DPO` output from the other LUT pair and creates a block to give vpr the information needed?

HackerFoo, 2019-04-23:
RAM128X1D is like RAM256X1S, except the 8 mux isn't used, instead using the outputs from both of the 7 muxes. This still takes a whole slice, but it makes the two halves independent, yet we'd still like both halves to be packed into the same slice.

To force VPR to do this, the outputs from both 7 muxes (SPO and DPO) are routed through a stub that is a black box to VPR, so now both halves can be combined into a single molecule (like a RAM256X1S) even though the outputs do not otherwise connect to a common atom.

mithro, 2019-04-25:
Yeah! That sounds about right.

Can you get this documentation into the repository somewhere?

Would be good to have a diagram with it. There should be comments in the
techmap and xml files pointing to this documentation too.

I would also rename stub to something like "FORCE_ASSOCIATION" or maybe
"FORCE_COMBINED_PLACEMENT"?

litghost, 2019-04-29:
I would generalize that all LUT-RAM dual ports could be packed various ways, but to ensure that both halves of the LUT-RAM reside in the SLICE and are adjacent, the black box is used like @HackerFoo said.