f4pga/f4pga-arch-defs #626: Clean up the top level directory

mithro, 2019-04-23:
We should try and keep the top level directory of the repository relatively clean.

I suggest we do the following;
 - [ ] Move `make` to `common/cmake`
 - [ ] Move `yosys` to `common/yosys`
 - [ ] Move `library` to `common/verilog`
 - [ ] Move `cmake.md` to `docs`

Maybe we should rename `common` to `support`?

This leaves us with the following top level things;
 - `build`
 - `CMakeLists.txt`
 - `common`
 - `COPYING`
 - `docs`
 - `ecp5`
 - `ice40`
 - `Makefile`
 - `README.md`
 - `testarch`
 - `tests`
 - `third_party`
 - `utils`
 - `vpr`
 - `xc7`

mithro, 2019-04-23:
@elmsfu - Thoughts?

mithro, 2019-04-23:
My general aim is to have as close as possible to one top level directory for each architecture.

mithro, 2019-04-23:
Issue #571 is part of this.

elmsfu, 2019-04-23:
I like the suggestions.

A few others to consider
 - [ ] put all archs in a subdirectory 
 - [ ] move `tests` to `common/test_designs`

mithro, 2019-04-23:
In some ways I think a wire isn't really a "design"....

mithro, 2019-04-23:
I kind of like having the architectures at the top level directory...

elmsfu, 2019-04-23:
I can see either way on archs, but I think `tests` is a poor name and is intended to be common

mithro, 2019-04-23:
Yeah, I agree to that...