f4pga/f4pga-arch-defs #640: Make sure IS_WCLK_INVERTED is passed for all DRAM types

HackerFoo, 2019-04-24:
https://github.com/SymbiFlow/symbiflow-arch-defs/blob/ec20de8cef15e087478a7f78485f27105eabc90e/xc7/techmap/cells_map.v#L497

mithro, 2019-04-25:
@HackerFoo I'm not sure what this issue is describing?

mithro, 2019-04-25:
Is the issue that `.IS_WCLK_INVERTED` is hard coded to `1'b0` rather than using the `localparam` value?

mkurc-ant, 2019-04-25:
@HackerFoo
I did some looking, here are my findings:

As for the most recent Yosys (master+wip @2c7e2541) the RAM32X1D is not included in techmap at all. It has to be defined there:
- `techlibs/xilinx/drams.txt`
- `techlibs/xilinx/drams_map.txt`

So until you instantiate it explicitly, Yosys will not output this primitive.

There are no FASM features related to clock inversion for any of SPRAMs and
DPRAMs. So it seems that the parameter is ignored for VPR anyways.

Definitions of SPRAMs and DPRAMs for VPR can be found there in arch-defs:
- `xc7/primitives/slicem/Ndram/`

Moreover there are no bits related to the clock inversion for DRAMs. See `segbits_clblm_l.db` in `prjxray-db/artix7` and there is a reason for that:

When you look at the schematic of the SLICEM in Vivado you will see that there is only one inverter for the clock and it is SLICE wide.

I think for now we are not taking advantage of that feature in VPR. In order to use it the VPR need to know which primitives do require clock inversion and would have to pack them into the same SLICE. And then enable the clock inversion for the whole SLICE.

Anyways you are right, the `.IS_WCLK_INVERTED` should not be forced to `1'b0`

mkurc-ant, 2019-04-25:
I made a PR: https://github.com/SymbiFlow/symbiflow-arch-defs/pull/649

mkurc-ant, 2019-04-25:
[image]

closed 2019-04-29 by HackerFoo