f4pga/f4pga-arch-defs #642: Make v2x produce the new composable interconnect specification again (IE Revert #637).

mithro, 2019-04-24:
The pull request https://github.com/SymbiFlow/symbiflow-arch-defs/pull/637 changed v2x to output the **old** interconnect format. This we should probably revert #637.

> This is the **old** format currently used by upstream VtR
> ```
> <direct input="<current_module>.<input>" name="<name>" output="<child_module>.<output>"/>
> <direct input="<child_module>.<input>" name="<name>" output="<current_module>.<output>"/>
> ```
> 
> This is the **new** format;
> ```
> <direct>
>   <port name="<input>" type="input"/>
>   <port from="<child_module>" name="<output>" type="output"/>
> </direct>
> <direct>
>   <port from="<child_module>" name="<input>" type="input"/>
>   <port name="<output>" type="output"/>
> </direct>
> ```
> 
> Notice how in the current VtR format the **current** module name and the **child** module name need to specified. In the new VtR format only the **child** name needs to be specified.
> 
> The new format enables the following pattern;
> ```
>  <pb_type name="module1" num_pb="1">
>   <xi:include href="module.xml" xpointer="xpointer(pb_type/child::node())" />
>  </pb_type>
>  <pb_type name="module2" num_pb="1">
>   <xi:include href="module.xml" xpointer="xpointer(pb_type/child::node())" />
>  </pb_type>
> ```
> IE You can use an `xi:include` to import the same module under two different names.
> 
> I go through some of my ideas around making the VtR XML format more composible in the VtR issue https://github.com/verilog-to-routing/vtr-verilog-to-routing/issues/272
> 
> As VtR doesn't actually support the new format, we use a xsltproc script to translate the XML file into the old format at architecture merging time. See this part of [xmlsort.xsl](https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/common/xml/xmlsort.xsl#L39-L51);
> https://github.com/SymbiFlow/symbiflow-arch-defs/blob/2a41dcc11eee49c3315d811e7060409c4a9fac56/common/xml/xmlsort.xsl#L39-L51
> 

I assume that #637 was merged because other pull requests depend on it. Hence I'm logging this issue to understand what needs to be done.

mithro, 2019-04-24:
@kgugala + @acomodi - FYI

mithro, 2019-05-07:
@acomodi This is done right?

acomodi, 2019-05-08:
@mithro Yes, closing it. fixed by https://github.com/SymbiFlow/symbiflow-arch-defs/pull/648

closed 2019-05-08 by acomodi