f4pga/f4pga-arch-defs #667: vlog_to_model does not recognize clock assignment if a port is not connected directly to $dff

kgugala, 2019-05-01:
in the following case:

```verilog
module simple_dff_logic(clk, a, b, q)
    input wire clk;
    input wire a;
    input wire b;
    output reg q;

    always @ (posedge clk) begin
        q <= a | b;
    end
endmodule
```

The $dff input is connected to $or output, `a` and `b` signals are connected to the $or inputs.

the `is_registered` function (see: https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/utils/vlog/vlog_to_model.py#L29) checks only $dff inputs, so in the above example `a` and `b` will not be classified as assigned to `clk` clock.

mithro, 2019-05-01:
Looking at these functions;
https://github.com/SymbiFlow/symbiflow-arch-defs/blob/dc45861228a81ade14f22d828856a575f9b471d9/utils/vlog/vlog_to_model.py#L29-L73

They should be using Yosys to find this information. Yosys has the ability to trace through the logic.

mithro, 2019-05-01:
The issue is likely in;

https://github.com/SymbiFlow/symbiflow-arch-defs/blob/dc45861228a81ade14f22d828856a575f9b471d9/utils/vlog/vlog_to_model.py#L195-L209

kgugala, 2019-05-02:
This commit https://github.com/SymbiFlow/symbiflow-arch-defs/pull/664/commits/0349a652be87a5b5b08c3ce2d38b545c7c94b9d2

seems to be fixing this. It is a part of #664

kgugala, 2019-05-08:
Since fix has been merged, we can close this issue

closed 2019-05-08 by kgugala