f4pga/f4pga-arch-defs #690: Carrychain specialization and Round Robin packing

acomodi, 2019-05-07:
With the tile split merged (https://github.com/SymbiFlow/symbiflow-arch-defs/pull/665) VPR now works at the `SLICE` level instead of `CLB`.

We need to check and make sure whether the carry-chain specialization is still required, as well as Round Robin.

- Carry-chain specialization: it assigns a unique name to the carry-chain pack-pattern to force the packer to pack all the carry-chains in a macro (without this the placer would place parts of the carry-chain in incorrect locations);
- Round Robin packing: packs the macros in a round robin fashion. This will allow to use all of the possible tiles to implement a carry-chain (right now the `first fit` algorithm is used to select the tile)

litghost, 2019-05-07:
There are multiple aspects to this issue:

- Will VPR work at all with specialization disabled? 
- What is the quality of results with specialization enabled vs disabled?  This is about both resource utilization and timing.

litghost, 2020-10-29:
This issue is very old

closed 2020-10-29 by litghost