f4pga/f4pga-arch-defs #725: Make vpr_pbtype_to_eblif support inbuilt blif model types

mithro, 2019-05-14:
The vpr_pbtype_to_eblif should support the inbuilt blif model types. The [complete list is found here](https://vtr-verilog-to-routing.readthedocs.io/en/latest/arch/reference.html#pb-type) and listed below;

 - [x] N/A - `.input`: A BLIF netlist input
 - [x] N/A - `.output`: A BLIF netlist output
 - [ ] .names: A BLIF .names (LUT) primitive
 - [ ] .latch: A BLIF .latch (DFF) primitive
 - [x] .subckt <custom_type>: A user defined black-box primitive.

mithro, 2019-05-14:
@mkurc-ant I believe you already had a fix for this in https://github.com/SymbiFlow/symbiflow-arch-defs/pull/685 ?

mkurc-ant, 2019-05-15:
@mithro Actually no. I haven't touch the vpr_pbtype_to_eblif

mithro, 2019-05-16:
@mkurc-ant What about https://github.com/SymbiFlow/symbiflow-arch-defs/pull/685/commits/86e6af55747f95e229a6aaae4038f228ded98316 ?