f4pga/f4pga-arch-defs #736: Cmake warnings:

kgugala, 2019-05-20:
Fresh clone throws the following warings on `build env`:

```
CMake Warning at xc7/make/vivado.cmake:39 (message):
  Vivado targets for dram_test_64x1d_vivado not emitted, XRAY_VIVADO_SETTINGS
  env var must be set to point to Vivado settings.sh
Call Stack (most recent call first):
  xc7/tests/dram_test/CMakeLists.txt:30 (add_vivado_target)

CMake Warning at xc7/make/vivado.cmake:39 (message):
  Vivado targets for dram_test_32m_vivado not emitted, XRAY_VIVADO_SETTINGS
  env var must be set to point to Vivado settings.sh
Call Stack (most recent call first):
  xc7/tests/dram_test/CMakeLists.txt:30 (add_vivado_target)

CMake Warning at xc7/make/vivado.cmake:39 (message):
  Vivado targets for dram_test_64m_vivado not emitted, XRAY_VIVADO_SETTINGS
  env var must be set to point to Vivado settings.sh
Call Stack (most recent call first):
  xc7/tests/dram_test/CMakeLists.txt:30 (add_vivado_target)
```

acomodi, 2019-05-20:
@kgugala I had the same, to solve this you should set the env var `XRAY_VIVADO_SETTINGS` and the warnings won't show up again (https://github.com/SymbiFlow/symbiflow-arch-defs/blob/055447a13d508e362bad6fd2092febb4d36b849a/xc7/make/vivado.cmake#L39).

mithro, 2019-05-20:
We should probably add a default value which is the default Xilinx location.

litghost, 2019-05-20:
> We should probably add a default value which is the default Xilinx location.

There is no such thing as the default Xilinx location, and there is not guarantee it is even installed.

mithro, 2019-05-21:
Sure there is.

<hr>

### From prjxray

https://github.com/SymbiFlow/prjxray/blob/7ea41e257d9641d5633bb13eb276f5b48522258f/utils/environment.sh#L39

```
export XRAY_VIVADO_SETTINGS="${XRAY_VIVADO_SETTINGS:-/opt/Xilinx/Vivado/2017.2/settings64.sh}"
```

<hr>

### From LiteX BuildEnv

https://github.com/timvideos/litex-buildenv/blob/7fc049aad997d8deefb5980fc559fce5f2170675/scripts/enter-env.sh#L159-L197
```
                XILINX_SETTINGS_ISE='/opt/Xilinx/*/ISE_DS/settings64.sh'
                XILINX_SETTINGS_VIVADO='/opt/Xilinx/Vivado/*/settings64.sh'
```