f4pga/f4pga-arch-defs #748: Add v2x to use specify section rather than the extra timing attributes

mithro, 2019-05-21:
In the pull request https://github.com/SymbiFlow/symbiflow-arch-defs/pull/730/files#diff-9eaec90618c3dbb25af035f62cbe100b @litghost added timing definitions for the [`CARRY4_VPR` module](https://github.com/litghost/symbiflow-arch-defs/blob/b418932dc31a18fc91bf35825a80b2b965ccda95/xc7/primitives/common_slice/carry/carry4_vpr.sim.v). 

Which looks like the following;
```verilog
(* whitebox *)
module CARRY4_VPR(O0, O1, O2, O3, CO_CHAIN, CO_FABRIC0, CO_FABRIC1, CO_FABRIC2, CO_FABRIC3, CYINIT, CIN, DI0, DI1, DI2, DI3, S0, S1, S2, S3);
  parameter CYINIT_AX = 1'b0;
  parameter CYINIT_C0 = 1'b0;
  parameter CYINIT_C1 = 1'b0;

  (* DELAY_CONST_CYINIT="0.491e-9" *)
  (* DELAY_CONST_CIN="0.235e-9" *)
  (* DELAY_CONST_S0="0.223e-9" *)
  output wire O0;

  (* DELAY_CONST_CYINIT="0.613e-9" *)
  (* DELAY_CONST_CIN="0.348e-9" *)
  (* DELAY_CONST_S0="0.400e-9" *)
  (* DELAY_CONST_S1="0.205e-9" *)
  (* DELAY_CONST_DI0="0.337e-9" *)
  output wire O1;

  (* DELAY_CONST_CYINIT="0.600e-9" *)
  (* DELAY_CONST_CIN="0.256e-9" *)
  (* DELAY_CONST_S0="0.523e-9" *)
  (* DELAY_CONST_S1="0.558e-9" *)
  (* DELAY_CONST_S2="0.226e-9" *)
  (* DELAY_CONST_DI0="0.486e-9" *)
  (* DELAY_CONST_DI1="0.471e-9" *)
  output wire O2;
```

Yosys recently added support for the [Verilog `specify` tags](https://www.hdlworks.com/hdl_corner/verilog_ref/items/Specify.htm) (see https://github.com/YosysHQ/yosys/pull/946). This looks like;

```verilog
module test(EN, CLK, D, Q);
  wire [3:0] _0_;
  input CLK;
  input [3:0] D;
  input EN;
  output [3:0] Q;
  reg [3:0] Q;
  always @(posedge CLK)
      Q <= _0_;
  assign _0_ = EN ? D : Q;
  specify
    if (EN) (CLK *> (Q : D)) = (1:1:1, 2:3:4);
  endspecify
endmodule
```
And the following example;
```verilog
specify 
  specparam T1 = 3:4:5,
            T2 = 2:3:4;
  (A => Y) = (T1, T2);                 // Simple path
  (posedge Clk *> Q +: D) = (3, 5);    // Edge sensitive path
  if (Exp = 2'b00) (X *> Z) = 4;
  if (Exp = 2'b01) (X *> Z) = 3.5;
  ifnone (X *> Z) = 3;                 // State-dependent path
  pulsestyle_ondetect Q;               // Verilog-2001
  (D => Q) = (4, 6);
  showcancelled S;                     // Verilog-2001
  (R => S) = (2, 3)
  $setup(D, posedge Clk, 2.8);         // System timing check
endspecify 
```

It would be good if v2x could use the values from the specify section rather than the attributes.

mithro, 2019-05-21:
@mkurc-ant One solution might be to use the [Yosys Python Pass API](https://github.com/YosysHQ/yosys/tree/master/examples/python-api) to loop over the module and write out an SDF file and then add SDF loading to the v2x tool.

mithro, 2019-05-21:
Actually, maybe we should look at adding SDF loading to v2x first?

mkurc-ant, 2019-05-23:
@mithro It seems that Yosys currently does not support outputting "specify" timings though the JSON backend. I checked that using the following example:

```
module example(a, b, y);

 input wire a;
 input wire b;
 
 output wire y;

 assign y = a | b;
 
 specify
  specparam T1 = 1:2:3;
  specparam T2 = 4:5:6;

  (a => y) = (T1, T2);
  (b => y) = (T1, T2);

 endspecify
endmodule
```

And the JSON output is:
```
{
  "creator": "Yosys 0.8+486 (git sha1 5c164d08, gcc 7.4.0-1ubuntu1~18.04 -fPIC -Os)",
  "modules": {
    "example": {
      "attributes": {
        "cells_not_processed": 1,
        "src": "example.v:1"
      },
      "ports": {
        "a": {
          "direction": "input",
          "bits": [ 2 ]
        },
        "b": {
          "direction": "input",
          "bits": [ 3 ]
        },
        "y": {
          "direction": "output",
          "bits": [ 4 ]
        }
      },
      "cells": {
        "$or$example.v:8$1": {
          "hide_name": 1,
          "type": "$or",
          "parameters": {
            "A_SIGNED": 0,
            "A_WIDTH": 1,
            "B_SIGNED": 0,
            "B_WIDTH": 1,
            "Y_WIDTH": 1
          },
          "attributes": {
            "src": "example.v:8"
          },
          "port_directions": {
            "A": "input",
            "B": "input",
            "Y": "output"
          },
          "connections": {
            "A": [ 2 ],
            "B": [ 3 ],
            "Y": [ 4 ]
          }
        }
      },
      "netnames": {
        "$or$example.v:8$1_Y": {
          "hide_name": 1,
          "bits": [ 4 ],
          "attributes": {
            "src": "example.v:8"
          }
        },
        "a": {
          "hide_name": 0,
          "bits": [ 2 ],
          "attributes": {
            "src": "example.v:3"
          }
        },
        "b": {
          "hide_name": 0,
          "bits": [ 3 ],
          "attributes": {
            "src": "example.v:4"
          }
        },
        "y": {
          "hide_name": 0,
          "bits": [ 4 ],
          "attributes": {
            "src": "example.v:6"
          }
        }
      }
    }
  }
}
```

mkurc-ant, 2019-05-23:
@mithro False alarm: In order for Yosys to parse `specify` statements you need to explicitly enable it via `read_verilog -specify ...`. And then timings appear in JSON output. However it seems that the `specparam` statement is not supported...

Input:
```
module example(a, b, y);
 input wire a;
 input wire b;

 output wire y;

 assign y = a | b;
 
 specify

  (a => y) = (1:2:3, 4:5:6);
  (b => y) = (1:2:3, 4:5:6);

 endspecify
endmodule
```

Output:
```
{
  "creator": "Yosys 0.8+486 (git sha1 5c164d08, gcc 7.4.0-1ubuntu1~18.04 -fPIC -Os)",
  "modules": {
    "example": {
      "attributes": {
        "cells_not_processed": 1,
        "src": "example.v:1"
      },
      "parameters": {
      },
      "ports": {
        "a": {
          "direction": "input",
          "bits": [ 2 ]
        },
        "b": {
          "direction": "input",
          "bits": [ 3 ]
        },
        "y": {
          "direction": "output",
          "bits": [ 4 ]
        }
      },
      "cells": {
        "$or$example.v:8$3": {
          "hide_name": 1,
          "type": "$or",
          "parameters": {
            "A_SIGNED": 0,
            "A_WIDTH": 1,
            "B_SIGNED": 0,
            "B_WIDTH": 1,
            "Y_WIDTH": 1
          },
          "attributes": {
            "src": "example.v:8"
          },
          "port_directions": {
            "A": "input",
            "B": "input",
            "Y": "output"
          },
          "connections": {
            "A": [ 2 ],
            "B": [ 3 ],
            "Y": [ 4 ]
          }
        },
        "$specify$1": {
          "hide_name": 1,
          "type": "$specify2",
          "parameters": {
            "DST_WIDTH": 1,
            "FULL": 0,
            "SRC_DST_PEN": 0,
            "SRC_DST_POL": 0,
            "SRC_WIDTH": 1,
            "T_FALL_MAX": 6,
            "T_FALL_MIN": 4,
            "T_FALL_TYP": 5,
            "T_RISE_MAX": 3,
            "T_RISE_MIN": 1,
            "T_RISE_TYP": 2
          },
          "attributes": {
            "module_not_derived": 1,
            "src": "example.v:12"
          },
          "port_directions": {
            "DST": "input",
            "EN": "input",
            "SRC": "input"
          },
          "connections": {
            "DST": [ 4 ],
            "EN": [ "1" ],
            "SRC": [ 2 ]
          }
        },
        "$specify$2": {
          "hide_name": 1,
          "type": "$specify2",
          "parameters": {
            "DST_WIDTH": 1,
            "FULL": 0,
            "SRC_DST_PEN": 0,
            "SRC_DST_POL": 0,
            "SRC_WIDTH": 1,
            "T_FALL_MAX": 6,
            "T_FALL_MIN": 4,
            "T_FALL_TYP": 5,
            "T_RISE_MAX": 3,
            "T_RISE_MIN": 1,
            "T_RISE_TYP": 2
          },
          "attributes": {
            "module_not_derived": 1,
            "src": "example.v:13"
          },
          "port_directions": {
            "DST": "input",
            "EN": "input",
            "SRC": "input"
          },
          "connections": {
            "DST": [ 4 ],
            "EN": [ "1" ],
            "SRC": [ 3 ]
          }
        }
      },
      "netnames": {
        "$or$example.v:8$3_Y": {
          "hide_name": 1,
          "bits": [ 4 ],
          "attributes": {
            "src": "example.v:8"
          }
        },
        "a": {
          "hide_name": 0,
          "bits": [ 2 ],
          "attributes": {
            "src": "example.v:3"
          }
        },
        "b": {
          "hide_name": 0,
          "bits": [ 3 ],
          "attributes": {
            "src": "example.v:4"
          }
        },
        "y": {
          "hide_name": 0,
          "bits": [ 4 ],
          "attributes": {
            "src": "example.v:6"
          }
        }
      }
    }
  }
}
```