f4pga/f4pga-arch-defs #749: Add support for loading timing information in v2x from SDF

mithro, 2019-05-21:
We currently have a [Python parser for loading SDF files](https://github.com/SymbiFlow/python-sdf-timing). v2x is written in Python.

It would be good if for v2x we could use SDF files rather than timing attributes like found in the [CARRY4_VPR module](https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/xc7/primitives/common_slice/carry/carry4_vpr.sim.v), see below;
```verilog
(* whitebox *)
module CARRY4_VPR(O0, O1, O2, O3, CO_CHAIN, CO_FABRIC0, CO_FABRIC1, CO_FABRIC2, CO_FABRIC3, CYINIT, CIN, DI0, DI1, DI2, DI3, S0, S1, S2, S3);
  parameter CYINIT_AX = 1'b0;
  parameter CYINIT_C0 = 1'b0;
  parameter CYINIT_C1 = 1'b0;

  (* DELAY_CONST_CYINIT="0.491e-9" *)
  (* DELAY_CONST_CIN="0.235e-9" *)
  (* DELAY_CONST_S0="0.223e-9" *)
  output wire O0;

  (* DELAY_CONST_CYINIT="0.613e-9" *)
  (* DELAY_CONST_CIN="0.348e-9" *)
  (* DELAY_CONST_S0="0.400e-9" *)
  (* DELAY_CONST_S1="0.205e-9" *)
  (* DELAY_CONST_DI0="0.337e-9" *)
  output wire O1;

  (* DELAY_CONST_CYINIT="0.600e-9" *)
  (* DELAY_CONST_CIN="0.256e-9" *)
  (* DELAY_CONST_S0="0.523e-9" *)
  (* DELAY_CONST_S1="0.558e-9" *)
  (* DELAY_CONST_S2="0.226e-9" *)
  (* DELAY_CONST_DI0="0.486e-9" *)
  (* DELAY_CONST_DI1="0.471e-9" *)
  output wire O2;
```

mithro, 2019-05-21:
This should probably be done before https://github.com/SymbiFlow/symbiflow-arch-defs/issues/748

mkurc-ant, 2019-05-23:
There is a PR: https://github.com/SymbiFlow/symbiflow-arch-defs/pull/765

litghost, 2020-03-02:
This has been resolved by patching timing after arch is complete.

closed 2020-03-02 by litghost