f4pga/f4pga-arch-defs #755: v2x VPR tests fail during packing stage

acomodi, 2019-05-22:
### Description
There is an issue with the v2x tests to check if the XML that is produced is VPR-compatible.
The VPR packing stage is not able to pack all the nets into one `TILE` only, but uses a `TILE` for each net, exceeding the available architecture resources. 

#### How does it work
v2x generated the `pb_type` and `model` XMLs of the architecture. `vpr_pbtype_arch_wrapper.py` encapsulates the `pb_type` in a complete architecture file.

At this point an `.eblif` file is produced to test VPR against the architecture file.

#### Setup
I have been testing this issue with the `multiple_instance` test. The architecture files and .eblif are the following:

Eblif
```
.model multiple_instance
.inputs a0 a1 b0 b1 cin0 cin1
.outputs cout0 cout1 sum0 sum1
.subckt adder a=a0 b=b0 cin=cin0 cout=cout0 sum=sum0
.subckt adder a=a1 b=b1 cin=cin1 cout=cout1 sum=sum1
.end
```

XML
```xml
<?xml version="1.0"?>
<!-- set: ai sw=1 ts=1 sta et -->
<architecture xmlns:xi="http://www.w3.org/2001/XInclude">
  <models>
    <model name="adder">
      <input_ports>
        <port combinational_sink_ports="cout sum" name="a"/>
        <port combinational_sink_ports="cout sum" name="b"/>
        <port combinational_sink_ports="cout sum" name="cin"/>
      </input_ports>
      <output_ports>
        <port name="cout"/>
        <port name="sum"/>
      </output_ports>
    </model>
  </models>
  <complexblocklist>
    <!-- input buffer -->
    <pb_type name="IBUF">
      <output name="I" num_pins="1"/>
      <pb_type blif_model=".input" name="PIN" num_pb="1">
        <output name="inpad" num_pins="1"/>
      </pb_type>
      <interconnect>
        <direct input="PIN.inpad" name="P" output="IBUF.I"/>
      </interconnect>
      <pinlocations pattern="custom">
        <loc side="right" xoffset="0" yoffset="0">IBUF.I</loc>
      </pinlocations>
    </pb_type>
    <!-- output buffer -->
    <pb_type name="OBUF">
      <input name="O" num_pins="1"/>
      <pb_type blif_model=".output" name="PIN" num_pb="1">
        <input name="outpad" num_pins="1"/>
      </pb_type>
      <interconnect>
        <direct input="OBUF.O" name="P" output="PIN.outpad"/>
      </interconnect>
      <pinlocations pattern="custom">
        <loc side="left" xoffset="0" yoffset="0">OBUF.O</loc>
      </pinlocations>
    </pb_type>
    <!-- pb_type.xml file will be included here -->
    <pb_type height="6" name="TILE" width="1">
      <pb_type name="multiple_instance" num_pb="1">
[... 166 lines cut ...]
Complex block 6: a0, type: IBUF .
Complex block 7: a1, type: IBUF .
Complex block 8: b0, type: IBUF .
Complex block 9: b1, type: IBUF .
Complex block 10: cin0, type: IBUF .
Complex block 11: cin1, type: IBUF .
	EMPTY: # blocks: 0, average # input + clock pins used: 0, average # output pins used: 0
	IBUF: # blocks: 6, average # input + clock pins used: 0, average # output pins used: 1
	OBUF: # blocks: 4, average # input + clock pins used: 1, average # output pins used: 0
	TILE: # blocks: 2, average # input + clock pins used: 3, average # output pins used: 2
Absorbed logical nets 0 out of 10 nets, 10 nets not absorbed.
FPGA sized to 8 x 8 (device)
Device Utilization: 0.34 (target 1.00)
	Block Utilization: 1.00 Type: IBUF
	Block Utilization: 1.00 Type: OBUF
	Block Utilization: 2.00 Type: TILE

# Packing took 0.03 seconds (max_rss 44.9 MiB, delta_rss +38.8 MiB)
Error 1: 
Type: Other
File: /tmp/really-really-really-really-really-really-really-really-really-really-really-really-really-long-path/conda/conda-bld/vtr_1557246610590/work/vpr/src/pack/pack.cpp
Line: 234
Message: Failed to find device which satisifies resource requirements required: IBUF: 6, OBUF: 4, TILE: 2 (available IBUF: 6, OBUF: 4, TILE: 1)
The entire flow of VPR took 0.04 seconds (max_rss 44.9 MiB)
```

litghost, 2019-05-22:
This is not a in VPR.  The grid generation script is generating a grid with only 1 TILE, see here:
```
      <single priority="1" type="IBUF" x="0" y="1"/>
      <single priority="1" type="IBUF" x="0" y="2"/>
      <single priority="1" type="IBUF" x="0" y="3"/>
      <single priority="1" type="IBUF" x="0" y="4"/>
      <single priority="1" type="IBUF" x="0" y="5"/>
      <single priority="1" type="IBUF" x="0" y="6"/>
      <single priority="1" type="TILE" x="2" y="1"/>
      <single priority="1" type="OBUF" x="4" y="2"/>
      <single priority="1" type="OBUF" x="4" y="3"/>
      <single priority="1" type="OBUF" x="4" y="4"/>
      <single priority="1" type="OBUF" x="4" y="5"/>
```

The bug is in https://github.com/SymbiFlow/symbiflow-arch-defs/blob/82bdc52f933fee25141b648b447c82bbde32f717/utils/vpr_pbtype_arch_wrapper.py#L93

acomodi, 2019-05-22:
@litghost The problem is that only one Tile should be used and not 2, in this case. I mean, VPR should pack everything in only one Tile because this one Tile contains two `adders` that should be used both instead of only one. 
I can also upload the packer `.net` output to have a clearer idea on what is happening

litghost, 2019-05-22:
> @litghost The problem is that only one Tile should be used and not 2, in this case. I mean, VPR should pack everything in only one Tile because this one Tile contains two `adders` that should be [...]

Okay, that is still not a VPR bug.  Was VPR told to pack as tightly as possible (pin utilization = 1 I believe)?  Because there are valid reasons not to, and the default doesn't target to consume all pins on clusters.

acomodi, 2019-05-22:
Yes, VPR is explicitly instructed to pack as densely as possible until saturation of the pins.

I believe there is something wrong in the architecture test file, but I still couldn't spot what the issue is. 
The eblif looks correct.

litghost, 2019-05-22:
I don't see "allow unrelated" enabled?

acomodi, 2019-05-22:
I have tried also to set unrelated clustering to `on` with no success though. Same error

mithro, 2019-05-23:
@acomodi When it failed to pack the first time, it reran itself with unrelated clustering to on. You should turn on the clustering debug.

mithro, 2019-05-23:
The options which might help;

 * `--connection_driven_clustering {on | off}` -- `off`
 * `--clustering_pin_feasibility_filter {on | off}`  -- `off`
 * `--pack_verbosity <int>`

```
vtr-verilog-to-routing/vpr/src/pack/cluster_router.cpp:/* #define PRINT_INTRA_LB_ROUTE */
```

litghost, 2019-09-17:
Didn't this get fixed?  Can we close this issue?  @acomodi @kgugala

acomodi, 2019-09-18:
@litghost Yes, can be closed. Fixed by https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/926

closed 2019-09-18 by acomodi