f4pga/f4pga-arch-defs #767: make test_python_utils fails on fresh checkout

mithro, 2019-05-23:
`make test_python_utils` fails if it is the only target you run from a fresh checkout. Looks like it is missing some dependency information?

See output below;
```
Downloading and Extracting Packages
py-1.8.0             | 65 KB     | ################################################################################################################################################### | 100%
attrs-19.1.0         | 32 KB     | ################################################################################################################################################### | 100%
more-itertools-4.3.0 | 83 KB     | ################################################################################################################################################### | 100%
pytest-4.5.0         | 354 KB    | ################################################################################################################################################### | 100%
pluggy-0.11.0        | 17 KB     | ################################################################################################################################################### | 100%
atomicwrites-1.3.0   | 9 KB      | ################################################################################################################################################### | 100%
wcwidth-0.1.7        | 17 KB     | ################################################################################################################################################### | 100%
Preparing transaction: done
Verifying transaction: done
Executing transaction: done
make[4]: Leaving directory '/usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs-v2x/build'                                                                                   
Built target conda_pytest
make[4]: Entering directory '/usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs-v2x/build'                                                                                  
Scanning dependencies of target test_python_utils
make[4]: Leaving directory '/usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs-v2x/build'                                                                                   
make[4]: Entering directory '/usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs-v2x/build'                                                                                  
==================================================================================== test session starts =====================================================================================
platform linux -- Python 3.7.1, pytest-4.5.0, py-1.8.0, pluggy-0.11.0 -- /usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs-v2x/build/env/conda/bin/python                  
cachedir: .pytest_cache
rootdir: /usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs-v2x/utils
collected 115 items / 6 errors / 109 selected                                                                                                                                                

=========================================================================================== ERRORS ===========================================================================================
______________________________________________________________________________ ERROR collecting clean_json5.py _______________________________________________________________________________
clean_json5.py:1: in <module>
    import pyjson5
E   ModuleNotFoundError: No module named 'pyjson5'
________________________________________________________________________ ERROR collecting lib/connection_database.py _________________________________________________________________________
lib/connection_database.py:3: in <module>
    from lib.rr_graph import graph2
lib/rr_graph/graph2.py:8: in <module>
    import progressbar
E   ModuleNotFoundError: No module named 'progressbar'
__________________________________________________________________________ ERROR collecting lib/rr_graph/graph2.py ___________________________________________________________________________
lib/rr_graph/graph2.py:8: in <module>
    import progressbar
E   ModuleNotFoundError: No module named 'progressbar'
_____________________________________________________________________ ERROR collecting lib/rr_graph/tests/test_graph2.py _____________________________________________________________________
lib/rr_graph/tests/test_graph2.py:5: in <module>
    from ..graph2 import SwitchTiming, SwitchSizing, Switch, SwitchType, \
lib/rr_graph/graph2.py:8: in <module>
    import progressbar
E   ModuleNotFoundError: No module named 'progressbar'
_____________________________________________________________________ ERROR collecting lib/rr_graph/tests/test_graph2.py _____________________________________________________________________
ImportError while importing test module '/usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs-v2x/utils/lib/rr_graph/tests/test_graph2.py'.                                   
Hint: make sure your test modules/packages have valid Python names.
Traceback:
../build/env/conda/lib/python3.7/site-packages/_pytest/python.py:512: in _importtestmodule
    mod = self.fspath.pyimport(ensuresyspath=importmode)
../build/env/conda/lib/python3.7/site-packages/py/_path/local.py:701: in pyimport
    __import__(modname)
../build/env/conda/lib/python3.7/site-packages/_pytest/assertion/rewrite.py:301: in load_module                                                                                              
    exec(co, mod.__dict__)
lib/rr_graph/tests/test_graph2.py:5: in <module>
    from ..graph2 import SwitchTiming, SwitchSizing, Switch, SwitchType, \
lib/rr_graph/graph2.py:8: in <module>
    import progressbar
E   ModuleNotFoundError: No module named 'progressbar'
________________________________________________________________________ ERROR collecting lib/rr_graph_xml/graph2.py _________________________________________________________________________
lib/rr_graph_xml/graph2.py:2: in <module>
    from lib.rr_graph import graph2
lib/rr_graph/graph2.py:8: in <module>
    import progressbar
E   ModuleNotFoundError: No module named 'progressbar'
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! Interrupted: 6 errors during collection !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
================================================================================== 6 error in 0.65 seconds ===================================================================================
utils/CMakeFiles/test_python_utils.dir/build.make:57: recipe for target 'utils/CMakeFiles/test_python_utils' failed                                                                          
make[4]: *** [utils/CMakeFiles/test_python_utils] Error 2
make[4]: Leaving directory '/usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs-v2x/build'                                                                                   
CMakeFiles/Makefile2:8334: recipe for target 'utils/CMakeFiles/test_python_utils.dir/all' failed                                                                                             
make[3]: *** [utils/CMakeFiles/test_python_utils.dir/all] Error 2
make[3]: Leaving directory '/usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs-v2x/build'                                                                                   
CMakeFiles/Makefile2:8341: recipe for target 'utils/CMakeFiles/test_python_utils.dir/rule' failed                                                                                            
make[2]: *** [utils/CMakeFiles/test_python_utils.dir/rule] Error 2
make[2]: Leaving directory '/usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs-v2x/build'                                                                                   
Makefile:2775: recipe for target 'test_python_utils' failed
make[1]: *** [test_python_utils] Error 2
make[1]: Leaving directory '/usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs-v2x/build'                                                                                   
Makefile:20: recipe for target 'test_python_utils' failed
make: *** [test_python_utils] Error 2
```

mithro, 2019-05-23:
It is unclear to me why this isn't failing on travis?

litghost, 2019-05-23:
> It is unclear to me why this isn't failing on travis?

There is a missing dependency from test_python on the conda python logic.  I explicitly fixed this on the xc7 python stuff: https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/xc7/CMakeLists.txt#L15

closed 2019-05-27 by mithro