f4pga/f4pga-arch-defs #780: xc7: ntemplate.nlut.sim.v and ntemplate.nlut.pb_type.xml don't match

mithro, 2019-05-26:
Currently the `ntemplate.nlut.sim.v` file has the following;
https://github.com/SymbiFlow/symbiflow-arch-defs/blob/26687ddc283a255ad804453d1f61e8f71e6f533a/xc7/primitives/common_slice/Nlut/ntemplate.Nlut.sim.v#L31-L34

While the `ntemplate.pb_type.xml` file has the following;
https://github.com/SymbiFlow/symbiflow-arch-defs/blob/26687ddc283a255ad804453d1f61e8f71e6f533a/xc7/primitives/common_slice/Nlut/ntemplate.Nlut.pb_type.xml#L65-L74

The `pb_type.xml` indicates a routing mux on the output of one of the `O6` line which selects between the second `LUT5` and the `F6MUX` output. However, this information is not being communicated in the bitstream as there is no fasm feature associated with this MUX?

Is there something special going on with the `SPLIT_LUT` line below?
https://github.com/SymbiFlow/symbiflow-arch-defs/blob/26687ddc283a255ad804453d1f61e8f71e6f533a/xc7/primitives/common_slice/Nlut/ntemplate.Nlut.pb_type.xml#L77

mithro, 2019-05-26:
I believe the relevant information might be in the comment here https://github.com/SymbiFlow/symbiflow-arch-defs/issues/619#issuecomment-487308569

mithro, 2019-05-26:
I think I understand what is going on here.

The two modes are;
 * (a) When the block is used as a `LUT6` there will be 2 * `LUT5` + a `F6MUX` and hence VtR will use one path of the O6 mux.
 * (b) When the block is used as 2 * `LUT5`, no `F6MUX` will exist, hence VtR will effectively "bypass" the `F6MUX` and make the `O6` output equal to the other LUT output. 

In case (b) the A6 pin should be connected to VCC. I don't see how this ends up in the FASM however?

mithro, 2019-05-26:
@litghost How does the A6 pin end up connected to VCC in case (b)?

litghost, 2019-05-28:
> @litghost How does the A6 pin end up connected to VCC in case (b)?

Implicitly.  A "disconnected" LUT pin is VCC.