f4pga/f4pga-arch-defs #788: Modeling the FDCE with timing for VPR

mkurc-ant, 2019-05-28:
Given the FDCE verilog model:

```
module fdce (DIN, CE, CLK, S, OUT);
 input  wire DIN;
 input  wire CE;
 input  wire CLK;
 (* SDF_ALIAS = "SR" *)
 input  wire S;
 (* SDF_ALIAS = "Q" *)
 output reg  OUT;

 always @(posedge CLK or posedge S)
  if      (S)   OUT <= 1'd0;
  else if (CE)  OUT <= DIN;

endmodule
```

I hand-wrote the model XML due to issue https://github.com/SymbiFlow/symbiflow-arch-defs/issues/784  with the V2X. And I generated the pb_type XML using my WIP version of the V2X which injects timings from SDF. The XML files look like that:

```
<models xmlns:xi="http://www.w3.org/2001/XInclude">
  <model name="fdce">
    <input_ports>
      <port clock="CLK"  name="CE"/>
      <port is_clock="1" name="CLK"/>
      <port clock="CLK"  name="DIN"/>
      <port clock="CLK" combinational_sink_ports="OUT" name="S"/>
    </input_ports>
    <output_ports>
      <port clock="CLK" name="OUT"/>
    </output_ports>
  </model>
</models>
```
```
<?xml version='1.0' encoding='utf-8'?>
<pb_type xmlns:xi="http://www.w3.org/2001/XInclude" blif_model=".subckt fdce" name="fdce" num_pb="1">
  <input name="CE" num_pins="1"/>
  <clock name="CLK" num_pins="1"/>
  <input name="DIN" num_pins="1"/>
  <output name="OUT" num_pins="1"/>
  <input name="S" num_pins="1"/>
  <T_clock_to_Q clock="CLK" port="OUT" min="2.440e-10" max="3.030e-10"/>
  <T_hold clock="CLK" port="CE" value="-9.000e-12"/>
  <T_hold clock="CLK" port="DIN" value="2.250e-10"/>
  <T_setup clock="CLK" port="CE" value="1.090e-10"/>
  <T_setup clock="CLK" port="DIN" value="-4.600e-11"/>
  <delay_constant in_port="S" out_port="OUT" min="6.160e-10" max="7.640e-10"/>
  <T_setup clock="CLK" port="S" value="4.040e-10"/>
  <T_hold clock="CLK" port="S" value="-2.480e-10"/>
</pb_type>
```
And the VPR complains that it does not have a timing defined:

`Error 1: /home/mkurc/Repos/google-symbiflow-arch-defs/build/utils/vlog/tests/timing_ff/fdce.arch.merged.xml:-1 <pb_type> 'fdce' timing-annotation/<model> mismatch on port 'S' of model 'fdce', port is a sequential input with internal combinational connects but has neither min nor max T_clock_to_Q specified`

So after poking around I manually added timings which the VPR required and ended up with the following pb_type XML (the model remains unchanged)

```
<?xml version='1.0' encoding='utf-8'?>
<pb_type xmlns:xi="http://www.w3.org/2001/XInclude" blif_model=".subckt fdce" name="fdce" num_pb="1">
  <input name="CE" num_pins="1"/>
  <clock name="CLK" num_pins="1"/>
  <input name="DIN" num_pins="1"/>
  <output name="OUT" num_pins="1"/>
  <input name="S" num_pins="1"/>
  <T_clock_to_Q clock="CLK" port="OUT" min="2.440e-10" max="3.030e-10"/>
  <T_hold clock="CLK" port="CE" value="-9.000e-12"/>
  <T_hold clock="CLK" port="DIN" value="2.250e-10"/>
  <T_setup clock="CLK" port="CE" value="1.090e-10"/>
  <T_setup clock="CLK" port="DIN" value="-4.600e-11"/>
  <delay_constant in_port="S" out_port="OUT" min="6.160e-10" max="7.640e-10"/>
  <T_setup clock="CLK" port="S" value="4.040e-10"/>
  <T_hold clock="CLK" port="S" value="-2.480e-10"/>

  <T_clock_to_Q clock="CLK" port="S" max="3e-10"/>
  <T_setup clock="CLK" port="OUT" value="1e-11"/>
  <T_hold clock="CLK" port="OUT" value="1e-11"/>
</pb_type>
```

The VPR accepted such architecture definition.

My questions are:
 - Is the model XML correct ?
 - What those additional timings actually represent ?
 - The additional timings are not present in SDF so what values (or tags) should i place there ?

litghost, 2019-05-28:
I believe http://docs.verilogtorouting.org/en/latest/tutorials/arch/timing_modeling/#mixed-sequential-combinational-block is the closest example.  And the error is pretty clear, you need a T_clk_to_Q from the SR pin to the output

litghost, 2019-05-28:
I think:
```
      <port clock="CLK" combinational_sink_ports="OUT" name="S"/>
```
Should be:
```
      <port combinational_sink_ports="OUT" name="S"/>
```

And you still specific T_hold/T_setup to model the recovery/release constraints.

mkurc-ant, 2019-05-28:
But the `T_clk_to_Q` tag should specify timing for a path from clock to output. I don't understand the meaning of timing specified like in the VPR example, it concerns clock and input:
```
<T_clock_to_Q max="200e-12" port="mem_sp.addr" clock="clk"/>
```

I did the `<port combinational_sink_ports="OUT" name="S"/>` change. And now the VPR complains that there are timings specified which consider CLK and S but the signal S is async according to the model (which makes sense).

The error refers to the line 58 in arch which is T_setup tag.
```
fdce.arch.merged.xml:58 <pb_type> timing-annotation/<model> mismatch on port 'S' of model 'fdce', model specifies no clock but timing annotation specifies 'CLK'
```

mkurc-ant, 2019-05-28:
@litghost I did another test. I copied model and pb_type from the VPR documentation and changed only names to "fdce". And the VPR fails with error:

```
fdce.arch.merged.xml:53 <pb_type> timing-annotation/<model> mismatch on port 'addr' of model 'fdce', model specifies no clock but timing annotation specifies 'clk'
```

The model is:
```
<model name="fdce">                                                         
  <input_ports>                                                             
    <port combinational_sink_ports="out" name="we"/>                        
    <port combinational_sink_ports="out" name="addr"/>                      
    <port combinational_sink_ports="out" name="data"/>                      
    <port is_clock="1" name="clk"/>                                         
  </input_ports>                                                            
  <output_ports>                                                            
    <port name="out"/>                                                      
  </output_ports>                                                           
</model> 
```

And the pb_type is:
```
<pb_type blif_model=".subckt fdce" name="fdce" num_pb="1">                
  <input name="addr" num_pins="9"/>                                       
  <input name="data" num_pins="64"/>                                      
  <input name="we" num_pins="1"/>                                         
  <output name="out" num_pins="64"/>                                      
  <clock name="clk" num_pins="1"/>                                        
  <!-- External input register timing -->                                 
  <T_setup clock="clk" port="mem_sp.addr" value="50e-12"/>                
  <T_setup clock="clk" port="mem_sp.data" value="50e-12"/>                
  <T_setup clock="clk" port="mem_sp.we" value="50e-12"/>                  
  <!-- Internal input register timing -->                                 
  <T_clock_to_Q clock="clk" max="200e-12" port="mem_sp.addr"/>            
  <T_clock_to_Q clock="clk" max="200e-12" port="mem_sp.data"/>            
  <T_clock_to_Q clock="clk" max="200e-12" port="mem_sp.we"/>              
  <!-- Internal combinational delay -->                                   
  <delay_constant in_port="mem_sp.addr" max="800e-12" out_port="mem_sp.out"/>
  <delay_constant in_port="mem_sp.data" max="800e-12" out_port="mem_sp.out"/>
  <delay_constant in_port="mem_sp.we" max="800e-12" out_port="mem_sp.out"/>
</pb_type> 
```

Maybe the rest of the architecture definition confuses the VPR...

litghost, 2019-05-28:
At this point I'd assume the documentation is broken.

I recommend we open an issue with upstream VTR to:
1. Add a way for their documentation cases to be tested.  I believe in this case the documentation example is simply broken
2. Determine what the best way forward is

Async resets on FF's are no uncommon features, so I assume there must be a way to do this.

mkurc-ant, 2019-05-28:
Probably it is broken. I'll prepare a more reproducible example and file the issue tomorrow.

mithro, 2019-05-29:
@mkurc-ant / @litghost - We should update the VtR documentation to include all of async flip flop, sync flip flop and latch examples in their tutorial. As @litghost mentioned these types are relatively common things found in FPGAs.

mkurc-ant, 2019-05-29:
@litghost @mithro I submitted an issue to the VtR: https://github.com/verilog-to-routing/vtr-verilog-to-routing/issues/617

mithro, 2019-05-29:
@mkurc-ant With kmurry's answer please send a pull request to the VtR docs with info on how to model these flip flop types as examples.

mkurc-ant, 2019-05-30:
@mithro But I am still confused how to model an async S/R flip-flop. From the kmurry's answer:

> At the moment VPR assume any sequential port (i.e. with a clock="") is defining a FF, and has no direct combination connections, but may connect to internal timing paths (as shown here).

It seems that we need to completely decouple SR and CLK inputs. and then define `delay_constant` from SR to Q plus `T_clock_to_Q` for CLK to Q. I already tried that but it seems that the VPR does not support such timing configuration.

For now the only solution is to have a FF in with no timing spec on the async input.

mkurc-ant, 2019-05-30:
I tried to decouple the CLR (or SR) input from Q and specify no timings for it. Now the VPR accepts such architecture:

```
<model name="fdce">
  <input_ports>                                                             
    <port name="CLR"/>                                                      
    <port is_clock="1" name="CLK"/>                                         
    <port clock="CLK" name="DIN"/>                                          
    <port clock="CLK" name="CE"/>                                           
  </input_ports>                                                            
  <output_ports>                                                            
    <port clock="CLK" name="Q"/>                                            
  </output_ports> 
</model>
```
```
<pb_type blif_model=".subckt fdce" name="fdce" num_pb="4">                
  <input name="CLR" num_pins="1"/>                                        
  <clock name="CLK" num_pins="1"/>                                        
  <input name="DIN" num_pins="1"/>                                        
  <input name="CE" num_pins="1"/>                                         
  <output name="Q" num_pins="1"/>                                         
  <T_setup port="fdce.CE"  clock="CLK" value="1e-10"/>                    
  <T_hold  port="fdce.CE"  clock="CLK" value="1e-10"/>                    
  <T_setup port="fdce.DIN" clock="CLK" value="1e-10"/>                    
  <T_hold  port="fdce.DIN" clock="CLK" value="1e-10"/>                    
  <T_clock_to_Q port="fdce.Q" clock="CLK" max="1e-10"/>                   
</pb_type> 
```

Though this may be not what we want.

mkurc-ant, 2019-05-30:
Then I managed to add recovery/release (as setup/hold) for the async CLR input. The VPR accepts that but it thinks that the CLR is a sequential input just like D and CE.

```
<model name="fdce">                                                         
  <input_ports>                                                             
    <port name="CLR" clock="CLK"/>             
    <port is_clock="1" name="CLK"/>                                         
    <port clock="CLK" name="DIN"/>                                          
    <port clock="CLK" name="CE"/>                                           
  </input_ports>                                                            
  <output_ports>                                                            
    <port clock="CLK" name="Q"/>                                            
  </output_ports>                                                           
</model> 
```

```
<pb_type blif_model=".subckt fdce" name="fdce" num_pb="1">                
  <input name="CLR" num_pins="1"/>                                        
  <clock name="CLK" num_pins="1"/>                                        
  <input name="DIN" num_pins="1"/>                                        
  <input name="CE" num_pins="1"/>                                         
  <output name="Q" num_pins="1"/>                                         
  <T_setup port="fdce.CE"  clock="CLK" value="1e-10"/>                    
  <T_hold  port="fdce.CE"  clock="CLK" value="1e-10"/>                    
  <T_setup port="fdce.DIN" clock="CLK" value="1e-10"/>                    
  <T_hold  port="fdce.DIN" clock="CLK" value="1e-10"/>                    
  <T_clock_to_Q port="fdce.Q" clock="CLK" max="1e-10"/>                   
  <T_setup port="fdce.CLR" clock="CLK" value="1e-10"/>                    
  <T_hold port="fdce.CLR" clock="CLK" value="1e-10"/>                     
</pb_type> 
```

[image]

And when I try to make the relation between CLR and Q combinational by adding `combinational_sink_ports="Q"` to CLR in the model then the VPR throws an error:

```
Error 1: ../arch.xml:-1 <pb_type> 'fdce' timing-annotation/<model> mismatch on port 'CLR' of model 'fdce', port is a sequential input with internal combinational connects but has neither min nor max T_clock_to_Q specified
```

For me it makes no sense to add `T_clock_to_Q` for CLR input thus it does not define async timing from one port to another.

mkurc-ant, 2019-05-30:
Now I realized that defining Trec/Trel for async reset makes no sense at all..

mkurc-ant, 2019-05-31:
I filed an issue to the upstream VPR with request for adding the ability to model async S/R flip-flop.