f4pga/f4pga-arch-defs #810: Xilinx tests failing with AttributeError: module 'prjxray.tile' has no attribute 'OutpinTiming'

mithro, 2019-06-05:
https://source.cloud.google.com/results/invocations/3344eb49-e07d-44c6-b30a-0287ba31bcc4/log
```
100% (2837 of 2837) |####################| Elapsed Time: 0:00:00 Time:  0:00:00
Loading database from '/tmpfs/src/github/symbiflow-arch-defs-presubmit-xc7/build/xc7/archs/artix7/channels.db'
2019-06-05 14:28:40.578516: About to load database
Dumping database to '/tmpfs/src/github/symbiflow-arch-defs-presubmit-xc7/build/xc7/archs/artix7/channels.db'
Traceback (most recent call last):
  File "/tmpfs/src/github/symbiflow-arch-defs-presubmit-xc7/xc7/utils/prjxray_form_channels.py", line 1651, in <module>
    main()
  File "/tmpfs/src/github/symbiflow-arch-defs-presubmit-xc7/xc7/utils/prjxray_form_channels.py", line 1630, in main
    import_phy_grid(db, grid, conn, get_switch, get_switch_timing)
  File "/tmpfs/src/github/symbiflow-arch-defs-presubmit-xc7/xc7/utils/prjxray_form_channels.py", line 390, in import_phy_grid
    db, write_cur, tile_types, site_types, tile_type, get_switch_timing
  File "/tmpfs/src/github/symbiflow-arch-defs-presubmit-xc7/xc7/utils/prjxray_form_channels.py", line 294, in add_wire_to_site_relation
    if isinstance(site_pin.timing, prjxray.tile.OutpinTiming):
AttributeError: module 'prjxray.tile' has no attribute 'OutpinTiming'
xc7/archs/artix7/CMakeFiles/file_xc7_archs_artix7_channels.db.dir/build.make:247: recipe for target 'xc7/archs/artix7/channels.db' failed
make[3]: *** [xc7/archs/artix7/channels.db] Error 1
make[3]: *** Deleting file 'xc7/archs/artix7/channels.db'
CMakeFiles/Makefile2:55746: recipe for target 'xc7/archs/artix7/CMakeFiles/file_xc7_archs_artix7_channels.db.dir/all' failed
make[2]: *** [xc7/archs/artix7/CMakeFiles/file_xc7_archs_artix7_channels.db.dir/all] Error 2
CMakeFiles/Makefile2:142: recipe for target 'CMakeFiles/all_xc7.dir/rule' failed
make[1]: *** [CMakeFiles/all_xc7.dir/rule] Error 2
Makefile:199: recipe for target 'all_xc7' failed
make: *** [all_xc7] Error 2
```

mithro, 2019-06-05:
I thought I saw something go by to fix this?

litghost, 2019-06-05:
https://github.com/SymbiFlow/symbiflow-arch-defs/pull/727/commits/0e52a41a83b19c0d5539108edd5f7175c9f6638d

litghost, 2019-06-05:
FYI, you broken master by merging https://github.com/SymbiFlow/symbiflow-arch-defs/pull/805

closed 2019-06-05 by litghost

litghost, 2019-06-05:
Also, with the router lookahead still behaving poorly has resulted in increased runtime: https://github.com/SymbiFlow/symbiflow-arch-defs/issues/716#issuecomment-499276865