f4pga/f4pga-arch-defs #816: Router lookahead issue tracking

litghost, 2019-06-06:
Currently the 7-series arch definition routing import and tile split is problematic for the router look ahead.  In the short term I'm planning to annotate the rr nodes with "canonical" locations and add support to https://github.com/SymbiFlow/vtr-verilog-to-routing/ to use these "canonical" locations.

In the long term, the canonical locations should be used for IPINs and as the locations used during lookahead map computations.  Several features or reworks are required to support this.

[ ] VPR needs to support tiles with hetrogenous cluster placements.  Currently VPR only supports the idea of 1 to N clusters at at a placement location that share the same cluster type.  For example, a CLBLL can be thought of as a SLICE_L, capacity=2.  However 7-series grids have several examples where the capacity style will not work, for example CLBLM.  Any time a tile contains more than one site (and site type), but connects to a single routing switch box this will come into play
[ ] Symbiflow-arch-defs needs to update IPIN locations to relate to the routing connection location
[ ] Symbiflow-arch-defs needs to generate channels that give "good" canonical coordinates.  This may require using more unidirectional channel types, rather than BIDIR.  Ultimately the goal is to allow VPR to compute "good" canonical coordinates

Once an initial lookahead is working, a test bench can be created to verify that the look ahead is working well, and as modifications are made to bring the VPR rr node description closer to the  canonical coordinates specified explicitly, lookahead performance should remain the same, or degrade slightly.

mithro, 2019-06-06:
@litghost Would the tile split idea help / affect this at all?

litghost, 2019-06-06:
> @litghost Would the tile split idea help / affect this at all?

Turns out the tile split **is** the problem.  We need to avoid the tile splits.  Instead we need to expose 1 tile with multiple clusters (which happen to be hetrogenous).  This heterogeneity is a perfect off shot of the equivalant ``<tile>`` tag.

litghost, 2019-06-06:
In general, the INT_LR tiles should be merged with their logic patterns in a way that preserves the routing grid structure (e.g. 1 grid unit is roughtly 1 unit of wire length).

closed 2019-06-12 by litghost