f4pga/f4pga-arch-defs #831: iCE40 BEL timing failure

elmsfu, 2019-06-14:
picosoc fails to route https://github.com/SymbiFlow/symbiflow-arch-defs/pull/693/commits/5c2fe5288273da1a98cb6816f9c19867486b935d

With no obvious cause I noticed that a minor change to one of the LUT timing
``` suggestion
<delay_constant in_port="LUT4.in[0]" max="4.48861e-10" out_port="LUT4.out"/>
<delay_constant in_port="LUT4.in[1]" max="3.99767e-10" out_port="LUT4.out"/>
- <delay_constant in_port="LUT4.in[2]" max="3.7872699999999997e-10" out_port="LUT4.out"/>
+ <delay_constant in_port="LUT4.in[2]" max="3.787269e-10" out_port="LUT4.out"/>
<delay_constant in_port="LUT4.in[3]" max="3.15606e-10" out_port="LUT4.out"/>
```
Allowed VPR to complete. Other minor changes also allowed for routing to complete.

acomodi, 2019-06-17:
@elmsfu I have been having a similar issue. picosoc is not routing for the ice40 (https://github.com/SymbiFlow/symbiflow-arch-defs/pull/830#issuecomment-502809633).

I have double-checked and c31cf57 revision of VTR is able to route picosoc for ice40 so there must be a regression between there and HEAD. I have started to check where the issue is.

elmsfu, 2019-06-18:
@acomodi I tried a few additional flags but no progress. [Here](https://gist.github.com/elmsfu/eb293de8ce3258d0bb0ced763a1c5666) are my logs from a run from the PR.

I started trying previous versions of VTR and will continue tomorrow.

acomodi, 2019-06-18:
@elmsfu The issue seems to be the same as the one I have encountered. 
I have identified the faulty commits (https://github.com/SymbiFlow/symbiflow-arch-defs/pull/830#issuecomment-503080786) and reverted them. I will push force on VTR master+wip a cleaned version with all the octopus merges on top of master as soon as the `all_xc7` and `all_ice40` pass locally.

acomodi, 2019-06-18:
[Here](https://github.com/SymbiFlow/vtr-verilog-to-routing/issues/108) instead, there is the issue linked to the revert.

elmsfu, 2019-06-20:
The above seemed to fix this for a while, but another change broke it again. The approach of adding vpr flag `--congested_routing_iteration_threshold 0.8` seemed to be more robust to allowing routing to succeed (see [here](https://github.com/SymbiFlow/symbiflow-arch-defs/pull/693#discussion_r295883606) ) .

closed 2019-06-20 by elmsfu