f4pga/f4pga-arch-defs #833: The XML schema for architectures needs to be updated

duck2, 2019-06-18:
This is supposed to be many issues, but I didn't want to open 7 of them at once, so I'm putting all schema bugs here.

### 1. X and Y offsets of metadata elements

Apparently, there are no X and Y offsets on metadata elements anymore, but the schema still accepts them.

```

###

mithro, 2019-06-18:
@duck2 - I think we would prefer multiple issues so it is easier to see progress?

elmsfu, 2019-06-18:
@duck2 We can create a symbiflow [project](https://github.com/orgs/SymbiFlow/projects) to show them on a single board if you'd like.

duck2, 2019-06-18:
ty for suggestions- I'll divide this issue up.

closed 2019-06-18 by duck2

mithro, 2019-06-19:
@duck2 - I created https://github.com/orgs/SymbiFlow/projects/15 for you.