f4pga/f4pga-arch-defs #841: Update the XML schema to not accept offsets on metadata

duck2, 2019-06-18:
Apparently, there are no X and Y offsets on metadata elements anymore, but the schema still accepts them:

```xsd
  <xs:complexType name="meta">
    <xs:simpleContent>
      <xs:extension base="xs:string">
        <xs:attribute name="name" type="xs:string" use="required"/>
        <xs:attribute name="xoffset" type="xs:int"/>
        <xs:attribute name="x_offset" type="xs:int"/>
        <xs:attribute name="yoffset" type="xs:int"/>
        <xs:attribute name="y_offset" type="xs:int"/>
      </xs:extension>
    </xs:simpleContent>
  </xs:complexType>
```

mithro, 2019-06-22:
@litghost Probably knows about what happened here?

litghost, 2019-06-25:
> @litghost Probably knows about what happened here?

The offset were never used in FASM, and upstream VTR devs identified that the offset logic was dead code, so it was removed.

mithro, 2019-09-10:
Fixed in #872.

closed 2019-09-10 by mithro