f4pga/f4pga-arch-defs #843: Disallow multiple input_ports in a model tag in the XML schema

duck2, 2019-06-18:
The VTR documentation says:

> Each model tag must contain 2 tags: <input_ports> and <output_ports>.

However, the schema and the [implementation](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/libs/libarchfpga/src/read_xml_arch_file.cpp#L2090) seems to be accepting multiple of them.

I'm not sure which one to take reference, docs or implementation?

```xsd
  <xs:complexType name="model_type">
    <xs:choice maxOccurs="unbounded">
      <xs:element name="input_ports" type="port_list"/>
      <xs:element name="output_ports" type="port_list"/>
    </xs:choice>
    <xs:attribute name="name" type="xs:string" use="required" />
  </xs:complexType>
```

mithro, 2019-06-22:
@duck2 - I think there should only be one `input_ports` and one `output_ports` and they both **must** exist.

mithro, 2019-09-10:
Fixed in #872.

closed 2019-09-10 by mithro