f4pga/f4pga-arch-defs #847: Update the switch type in the schema

duck2, 2019-06-18:
The docs and the implementation says the R, Cin, Cout... values in a switch type are floats, but the schema defines them as doubles.

The schema also defines a `Cinternal` attribute which isn't present anywhere in the code or the docs.

```xsd
  <xs:complexType name="switch">
    <xs:sequence>
      <xs:element name="Tdel" minOccurs="0" maxOccurs="unbounded">
        <xs:complexType>
          <xs:attribute name="num_inputs" type="xs:int" use="required"/>
          <xs:attribute name="delay" type="xs:double" use="required"/>
        </xs:complexType>
      </xs:element>
    </xs:sequence>
    <xs:attribute name="type" type="switch_type" use="required"/>
    <xs:attribute name="name" type="xs:string" use="required"/>
    <xs:attribute name="R" type="xs:double" use="required"/>
    <xs:attribute name="Cin" type="xs:double" use="required"/>
    <xs:attribute name="Cinternal" type="xs:double"/>
    <xs:attribute name="Cout" type="xs:double"/>
    <xs:attribute name="Tdel" type="xs:double"/>
    <xs:attribute name="buf_size" type="buf_size" default="auto"/>
    <xs:attribute name="mux_trans_size" type="xs:double"/>
    <xs:attribute name="power_buf_size" type="xs:double"/>
  </xs:complexType>
```

mithro, 2019-06-22:
`Cinternal` is in the process of being added. See https://github.com/verilog-to-routing/vtr-verilog-to-routing/issues/550 and https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/607

mithro, 2019-09-10:
Fixed in #872.

closed 2019-09-10 by mithro