f4pga/f4pga-arch-defs #848: Disallow multiple top-level tags in architecture tag

duck2, 2019-06-18:
The arch file reader implementation looks for one top level tag of each type:
```c++
        /* Process models */
        Next = get_single_child(architecture, "models", loc_data);
        [...]
```

The schema disagrees:
```xsd
  <xs:element name="architecture">
    <xs:complexType>
      <xs:choice minOccurs="0" maxOccurs="unbounded">
        <xs:element name="models" type="models"/>
        <xs:element name="layout" type="layout"/>
        <xs:element name="device" type="device"/>
        <xs:element name="switchlist" type="switchlist"/>
        <xs:element name="switchblocklist" type="switchblocklist"/>
        <xs:element name="segmentlist" type="segmentlist"/>
        <xs:element name="power" type="global_power"/>
        <xs:element name="clocks" type="clocks"/>
        <xs:element name="directlist" type="directlist"/>
        <xs:element name="tiles" type="tiles"/>
        <xs:element name="complexblocklist" type="complexblocklist"/>
      </xs:choice>
    </xs:complexType>
  </xs:element>
```

mithro, 2019-06-22:
You seem to be correct.

I think an architecture requires a lot of these `<xs:element>` objects?

mithro, 2019-09-10:
Fixed in #872.

closed 2019-09-10 by mithro