f4pga/f4pga-arch-defs #888: murax_basys test produces RAM64X1D

andrewb1999, 2019-07-18:
murax_basys test produces RAM64X1D that does not work when trying to load back into Vivado with toplevel_bit.v.tcl
[toplevel.eblif.log](https://github.com/SymbiFlow/symbiflow-arch-defs/files/3408351/toplevel.eblif.log)

andrewb1999, 2019-07-18:
Here is the bitstream, .v and .v.tcl
[murax_basys.tar.gz](https://github.com/SymbiFlow/symbiflow-arch-defs/files/3408362/murax_basys.tar.gz)

litghost, 2019-07-18:
RAM64X1D is fine, RAM64X1S is untested.  Synth log looks okay, checking out your bitstream.

litghost, 2019-07-18:
I can replicate the originally reported error, which was:
```
ERROR: [Vivado 12-2285] Cannot set LOC property of instance 'CLBLM_L_X10Y110_SLICE_X12Y110_RAM64X1S_C',  for bel C6LUT Conflicting nets for SLICE_X12Y110.C6LUT.WA4
Two net names are: CLBLL_L_X12Y109_SLICE_X16Y109_D_XOR and CLBLL_L_X14Y108_SLICE_X20Y108_DQ
Resolution: When using BEL constraints, ensure the BEL constraints are defined before the LOC constraints to avoid conflicts at a given site.
```

However I can see what the problem is via the FASM:
```
CLBLM_L_X10Y110.SLICEM_X0.ALUT.DI1MUX.BDI1_BMC31
CLBLM_L_X10Y110.SLICEM_X0.ALUT.RAM
CLBLM_L_X10Y110.SLICEM_X0.BLUT.DI1MUX.BI
CLBLM_L_X10Y110.SLICEM_X0.BLUT.RAM
CLBLM_L_X10Y110.SLICEM_X0.CLUT.DI1MUX.CI
CLBLM_L_X10Y110.SLICEM_X0.CLUT.RAM
CLBLM_L_X10Y110.SLICEM_X0.DLUT.RAM
```

It is a problem that ``CLBLM_L_X10Y110.SLICEM_X0.CLUT.DI1MUX.CI`` was emitted.  Unclear why that would be.

To debug further, please attach the arch.xml, rrgraph.xml, ``.net``, ``.place``, and ``.route`` files.  This could be latent bug in either VPR's FASM emission, or an arch definition error.  Unclear right now.

andrewb1999, 2019-07-18:
Here is everything but the rr_graph
[murax_basys_more.tar.gz](https://github.com/SymbiFlow/symbiflow-arch-defs/files/3408450/murax_basys_more.tar.gz)

andrewb1999, 2019-07-18:
Split rr_graph

[xaa.zip](https://github.com/SymbiFlow/symbiflow-arch-defs/files/3408483/xaa.zip)
[xab.zip](https://github.com/SymbiFlow/symbiflow-arch-defs/files/3408484/xab.zip)
[xac.zip](https://github.com/SymbiFlow/symbiflow-arch-defs/files/3408485/xac.zip)
[xad.zip](https://github.com/SymbiFlow/symbiflow-arch-defs/files/3408486/xad.zip)
[xae.zip](https://github.com/SymbiFlow/symbiflow-arch-defs/files/3408487/xae.zip)

litghost, 2019-07-18:
I can see in the ``.net`` that the packer choose to use the ``CLUT.DI1MUX.CI``, but this behavior is unexpected given that the ``CLUT.DI1MUX.DI`` net was already the correct net.  By choosing to change the mux to the ``CI`` port, an additional sink needed to be routed for the ``murax.system_apbBridge._zz_6[4]`` net.  I've never seen the VPR packer behavior this way (modify a mux when not required and add a sink to a net).  This likely constitutes a VPR bug, I'll need to consult with the VPR devs.

Which version of VPR did you use to pack murax?

litghost, 2019-07-18:
Here is a version of murax with working Verilog and TCL, and a Vivado DCP: [good_murax.tar.gz](https://github.com/SymbiFlow/symbiflow-arch-defs/files/3408663/good_murax.tar.gz)

litghost, 2019-07-18:
@andrewb1999: Please run murax_basys_route_echo and attach the ``pre_pack ... echo`` file.

andrewb1999, 2019-07-19:
@litghost 
Using VPR version from conda:
VPR FPGA Placement and Routing.
Version: 8.0.0-rc1+52120acea
Revision: vpr-7.0.5-7559-g52120acea

Here is the pre_pack echo:

[pre_packing_molecules_and_patterns.echo.tar.gz](https://github.com/SymbiFlow/symbiflow-arch-defs/files/3411768/pre_packing_molecules_and_patterns.echo.tar.gz)

closed 2020-09-30 by andrewb1999