f4pga/f4pga-arch-defs #926: [xc7/fasm2v] Fix decoding of slew rate

mkurc-ant, 2019-08-06:
In the database from `g50b9a02` slew rate features are connected with iostandards. Currently the fasm2v cannot decode them correctly and raises an exception:

```
Traceback (most recent call last):
  File "/usr/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/mkurc/Repos/google-symbiflow-arch-defs/xc7/fasm2bels/__main__.py", line 4, in <module>
    main()
  File "/home/mkurc/Repos/google-symbiflow-arch-defs/xc7/fasm2bels/fasm2bels.py", line 277, in main
    process_tile(top, tile, tile_features)
  File "/home/mkurc/Repos/google-symbiflow-arch-defs/xc7/fasm2bels/fasm2bels.py", line 80, in process_tile
    PROCESS_TILE[tile_type](top.conn, top, tile, tile_features)
  File "/home/mkurc/Repos/google-symbiflow-arch-defs/xc7/fasm2bels/iob_models.py", line 283, in process_iobs
    process_iob(top, iobs[iob])
  File "/home/mkurc/Repos/google-symbiflow-arch-defs/xc7/fasm2bels/iob_models.py", line 230, in process_iob
    add_output_parameters(bel, site)
  File "/home/mkurc/Repos/google-symbiflow-arch-defs/xc7/fasm2bels/iob_models.py", line 99, in add_output_parameters
    assert False
AssertionError
```

litghost, 2019-08-21:
This is fixed?

mkurc-ant, 2019-08-21:
@litghost Yes, fixed by https://github.com/SymbiFlow/symbiflow-arch-defs/pull/938

closed 2019-08-21 by mkurc-ant