f4pga/f4pga-arch-defs #929: [xc7] Build failure using most recent prjxray db (g50b9a02)

mkurc-ant, 2019-08-06:
When building bit file for `xc7/tests/counter` generation of the `channels.db` succeeds but the following error is thown afterwards:

```
100% (12964 of 12964) |########################################################################################################| Elapsed Time: 0:00:00 Time:  0:00:00
2019-08-06 14:27:35.914583 Flushing database back to file "/home/mkurc/Repos/google-symbiflow-arch-defs/build/xc7/archs/artix7/channels.db"
Built target file_xc7_archs_artix7_pin_assignments.json
Scanning dependencies of target file_xc7_archs_artix7_tiles_bram_l_bram_l.pb_type.xml
make[3]: *** No rule to make target '../third_party/prjxray-db/artix7/mask_bram_l.origin_info.db', needed by 'xc7/archs/artix7/tiles/bram_l/bram_l.pb_type.xml'.  Stop
```

It seems that the build system somehow considers file `../third_party/prjxray-db/artix7/mask_bram_l.origin_info.db` as a taget.

litghost, 2019-08-06:
Maybe re-run CMake build system?  I think this is a side effort of the glob.

mkurc-ant, 2019-08-12:
It seems that it has been solved by cleaning and re-running the build system.

closed 2019-08-12 by mkurc-ant