f4pga/f4pga-arch-defs #958: SRL32 missing timing

litghost, 2019-08-21:
https://github.com/SymbiFlow/symbiflow-arch-defs/blob/93541beb621030a30506312e86583ddcae5c327b/xc7/primitives/slicem/srl/srlc32e_vpr.pb_type.xml#L9

Timing values look like dummy values.

mkurc-ant, 2019-08-22:
Yep, they are dummy

mkurc-ant, 2019-08-22:
I started defining those timings and stumbled upon an issue. The SRL has output O6 which is sequentially related to inputs DI1,CE and combinationaly related to address input A. Like in the picture below:

[image]

Unfortunatelly such a structure is not supported by the VPR timing model. Supported structures are only those: http://docs.verilogtorouting.org/en/latest/tutorials/arch/timing_modeling/#arch-model-timing-tutorial

mkurc-ant, 2019-08-22:
https://github.com/SymbiFlow/symbiflow-arch-defs/pull/959

litghost, 2019-09-13:
Can this be closed?

mkurc-ant, 2019-09-16:
@litghost Yep.

closed 2019-10-30 by mkurc-ant