f4pga/f4pga-arch-defs #975: [xc7] Race condition on parallel build of `all_xc7_tests`

mkurc-ant, 2019-09-11:
When running `make all_xc7_tests -j 12` in a clean build directory (no conda libs downloaded yet) the following error is thrown:

```
Traceback (most recent call last):
  File "/home/build/mkurc/google-symbiflow-arch-defs/utils/mux_gen.py", line 12, in <module>
    import lxml.etree as ET
ImportError: libiconv.so.2: cannot open shared object file: No such file or directory
```

No such error shows up when building with `-j 1` or when re running make with `-j 12`. There has to be a race condition in with xc7 test targets and conda lib dependencies, namely `libiconv`.

litghost, 2019-09-13:
This particular race can be avoided by not trying to setup the conda env at the same time as the rest of the build.  The CI system runs:
```
make all_conda
make -j${MAX_CORES} all_xc7
```

It is worth noting on CI and GCS VM I have seen (but not been able to root cause) another racey looking error, which @acomodi mentioned [here](https://github.com/SymbiFlow/symbiflow-arch-defs/pull/987#pullrequestreview-287866285).  I have been unable to replicate this error on non-VM hardware, and I haven't been able to replicate this error when using Ninja either.  So unclear what is going on.

mkurc-ant, 2019-09-16:
You're right that it can be mitigated in the way the CI system does it. Nevertheless It's there, but I agree that it is not critical.

litghost, 2020-07-14:
I believe this can be closed now?

closed 2020-10-29 by litghost