f4pga/f4pga-arch-defs #996: [xc7] race condition in tests

kgugala, 2019-09-15:
when running `make all_xc7 -j40` the tests fail with:

```
SBY 10:55:55 [ff_ce_sr_17_fdpe_proof] Removing direcory 'ff_ce_sr_17_fdpe_proof'.                                                                              
SBY 10:55:55 [ff_ce_sr_17_fdpe_proof] Copy 'ff_ce_sr_17_fdpe/artix7-xc7a50t-basys3-roi-virt-xc7a50t-basys3-test/top_bit.v' to 'ff_ce_sr_17_fdpe_proof/src/top_bit.v'.
SBY 10:55:55 [ff_ce_sr_17_fdpe_proof] Copy 'ff_ce_sr_17_fdpe.v' to 'ff_ce_sr_17_fdpe_proof/src/ff_ce_sr_17_fdpe.v'.                                                                                                                                                                                                           
SBY 10:55:55 [ff_ce_sr_17_fdpe_proof] Copy 'ff_type.v' to 'ff_ce_sr_17_fdpe_proof/src/ff_type.v'.                                                                                                                                                                                                                             
SBY 10:55:55 [ff_ce_sr_17_fdpe_proof] engine_0: abc pdr                                                                                                        
SBY 10:55:55 [ff_ce_sr_17_fdpe_proof] nomem: starting process "cd ff_ce_sr_17_fdpe_proof/src; /symbiflow-arch-defs/build/env/conda/bin/yosys -ql ../model/design_nomem.log ../model/design_nomem.ys"
SBY 10:55:55 [ff_ce_sr_17_fdpe_proof] nomem: ERROR: No matching port in gate module was found for \led!                                                                                                                                                                                                                       
SBY 10:55:55 [ff_ce_sr_17_fdpe_proof] nomem: finished (returncode=1)                                                                                           
SBY 10:55:55 [ff_ce_sr_17_fdpe_proof] nomem: job failed. ERROR.                                                                                                
SBY 10:55:55 [ff_ce_sr_17_fdpe_proof] summary: Elapsed clock time [H:MM:SS (secs)]: 0:00:00 (0)                                                                                                                                                                                                                               
SBY 10:55:55 [ff_ce_sr_17_fdpe_proof] summary: Elapsed process time [H:MM:SS (secs)]: 0:00:00 (0)                                                                                                                                                                                                                             
SBY 10:55:55 [ff_ce_sr_17_fdpe_proof] DONE (ERROR, rc=16)
```

This is because `fasm2bels` produces empty `top_bit.v` (actually empty `top.bit.fasm` and then empty `top_bit.v`):
```
module top();
endmodule
```
The `fasm2bels'` input file (`top.bit`) is correct (at least when I checked it after the fail).
I rerun the tests a few times and they were always failing on one of the `ff_ce_sr*`

This looks like a race condition

kgugala, 2019-09-15:
@mithro @litghost PTAL

litghost, 2019-09-15:
I have replicated this issue when using the Makefile CMake driver + a VM.  I've not replicated on my own machine or on a VM using the Ninja CMake driver.  I cannot explain why Makefile + VM is different than the other conditions.

kgugala, 2019-09-15:
can we switch to ninja in kokoro?

litghost, 2019-09-15:
> can we switch to ninja in kokoro?

Yep, https://github.com/SymbiFlow/symbiflow-arch-defs/pull/991

litghost, 2020-03-02:
I believe this has been resolved

closed 2020-03-02 by litghost