f4pga/f4pga-arch-defs #1000: Convert to using sdc for Series 7 work

mithro, 2019-09-16:
Currently the Series 7 stuff uses `pcf` because the ice40 work used `pcf`. However it makes a lot more sense to use `sdc` as both vpr and vivado support that. (`xdc` is just a slightly custom version of `sdc`.)

litghost, 2019-09-16:
@HackerFoo / @mkurc-ant It likely makes sense to do this transition when breaking out of the ROI, as the things that are currently in the ROI harness (IOSTANDARD, DRIVE, etc) are typically specified as an xdc.

mkurc-ant, 2019-09-17:
I would very much like to have VPR to loc IOs according to `sdc` constraints. So there would be no need to parse `.net` to determine cluster names.

litghost, 2019-09-17:
> I would very much like to have VPR to loc IOs according to `sdc` constraints. So there would be no need to parse `.net` to determine cluster names.

The would require a change to VPR, VPR does not have `sdc` support for IO placement: https://docs.verilogtorouting.org/en/latest/vpr/sdc_commands/

mkurc-ant, 2019-09-17:
Still, I believe that such a modification is worth an effort. Because that would make the VPR complete as a P&R tool. Right now what we do is pack, process results using an external tool to add I/O constraints and then place and route. As an end user I'd like not to have many external tools, ideally just one that does the job.

vaughnbetz, 2019-10-03:
Agreed.  Hopefully #932 addresses this (when complete).