f4pga/f4pga-arch-defs #998: Add buttons_arty test

duck2, 2019-09-16:
I tried to add a buttons_arty test by adding an `arty` target to `xc7/tests/buttons` and creating the required `arty.pcf` and `buttons_arty.v` files, but building it failed with this error:

```
PCF constraint "set_io in[0] A8" from line 2 constraints pad A8 which is not in available pad map:
A9
C2
D10
E3
T10
```

This seems to because of the way the `arty` device is defined:
```
add_xc7_device_define_type(
...
  ROI_DIR ${PRJXRAY_DB_DIR}/artix7/harness/arty-a7/uart
...
)
```

In the prjxray repository, there is also `swbut` and `pmod` harnesses, but it doesn't seem to be possible to specify the harness in the test. How to go about this? Make a harness for arty including IO for both `uart` and `buttons` or try and generalize test targets?

acomodi, 2019-09-16:
@duck2 I believe the easiest way to add a buttons target for arty and use the `swbut` harness is to create another directory in `https://github.com/SymbiFlow/symbiflow-arch-defs/tree/master/xc7/archs/artix7/devices` and call it something like: "xc7a50t-arty-swbut-roi-virt". 

Than in there use the same [CMakeFile.txt](https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/xc7/archs/artix7/devices/xc7a50t-arty-roi-virt/CMakeLists.txt) as for the other arty device and change the ROI to be used.

acomodi, 2019-09-16:
Then, in the arty_button tests you will have to specify the correct I/Os in the arty.pcf files as they appear in the `swbut` ROI and add the arty_buttons target to the [CMakeLists.txt](https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/xc7/tests/buttons/CMakeLists.txt) using the newly added device

duck2, 2019-09-25:
#1006

closed 2019-09-25 by duck2