f4pga/f4pga-arch-defs #1065: BUFH route through

acomodi, 2019-10-10:
With https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1060 we are able to build a simple_ff test using the clocking network and ROI-less.

The next step is to get rid of the explicit BUFH instantiation that will prevent the whole 50T to have a valid clocking network. This because the BUFH is now placed in a specific location and cannot be accessed by other clock region.

The idea to solve this is to let VPR route through the BUFH tiles creating a CHAN to CHAN connection. Then with check_features add the relative feature of the BUFH that have been used during routing.

acomodi, 2019-10-10:
@litghost FYI

litghost, 2019-10-10:
Connecting the BUFH IPIN to OPIN is the simplest answer, but it won't work because of https://github.com/SymbiFlow/vtr-verilog-to-routing/blob/master%2Bwip/vpr/src/route/route_timing.cpp#L1859

Instead, a CHAN to CHAN (or CHAN to OPIN) edge is required.  Luckily there is a convenient graph node created during create edges [here](https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/xc7/utils/prjxray_edge_library.py#L347).  `find_wire_node` creates a CHAN node before each IPIN and a CHAN node after each OPIN that should be suitable for BUFH route through.

On another note, the prjxray-db already has a pip that represents the BUFH route through, but it has been ignored up till now because it is marked "is_psuedo", which represents that it is a site route through.  This is in fact what we are doing, so it can be used.  Example psuedo-pip for BUFH route through: https://raw.githubusercontent.com/SymbiFlow/prjxray-db/master/artix7/tile_type_CLK_HROW_BOT_R.json , search for CLK_HROW_BOT_R.CLK_HROW_CK_MUX_OUT_L0->>CLK_HROW_CK_HCLK_OUT_L0

Note that the pip has timing information representing the timing cost of routing through the BUFH.

acomodi, 2019-10-11:
@litghost I have tackled this issue today, but I still have some things I do not fully understand:

1) How the pseudo pip relative to the BUFH can be exploited to create the CHAN to CHAN connection? I have tried to not ignore this pip during edge creation (https://github.com/SymbiFlow/symbiflow-arch-defs/blob/90bf965ed4e47e8d59b958aa76741af404968a68/xc7/utils/prjxray_edge_library.py#L1766). I wanted to see what happened, and got to a point where the direction of the src and sink nodes were  inverted causing an assertion error (https://github.com/SymbiFlow/symbiflow-arch-defs/blob/90bf965ed4e47e8d59b958aa76741af404968a68/xc7/utils/prjxray_edge_library.py#L666).

2) Just to be completely sure, right now, without the pb_type, VPR is not able to route through BUFH because it does not know how to connect IPINs and OPINs inside the BUFH, correct? I mean, right now, if we do not explicitly insert BUFH, it is like VPR sees a blank spot at the place of the BUFH sites and the IPINs and OPINs of the CLK_HROW tile are hanging there without having a valid connection.

litghost, 2019-10-11:
> 1. How the pseudo pip relative to the BUFH can be exploited to create the CHAN to CHAN connection? I have tried to not ignore this pip during edge creation (

You'll need to make the connection directly, or add code to handle the psuedo pip directly, as it is "weird".  The existing code will not handle this case.

> 2\. Just to be completely sure, right now, without the pb_type, VPR is not able to route through BUFH

Kind of?  More accurately, the router routes from `SOURCES` to `SINKS`.  In the case of a BUFH route through, we want to avoid the IPIN/SINK/SOURCE/OPIN belonging to the BUFH altogether.

closed 2019-10-23 by litghost