f4pga/f4pga-arch-defs #1066: lxml generates exception when writing more than 2 GB incrementally

litghost, 2019-10-10:
There is a int32_t overflow bug in libxml2 (https://gitlab.gnome.org/GNOME/libxml2/issues/112) that causes an exception to be generated when more than 2 GB of XML is written.  The upstream maintainers have not merged a fix yet (see https://gitlab.gnome.org/GNOME/libxml2/merge_requests/54 or https://gitlab.gnome.org/GNOME/libxml2/merge_requests/55).  For now, libxml2 needs to be replaced.

Instructions:

TBD

litghost, 2019-10-10:
@mkurc-ant I'm working on creating a fork of https://github.com/conda-forge/libxml2-feedstock, I'll update the instruction once that is done and tested.

litghost, 2019-10-10:
@mkurc-ant Try out the instructions.  If you want to test if the fix is working, the following script should run without error if the fix is in place: [repo_lxml.py.zip](https://github.com/SymbiFlow/symbiflow-arch-defs/files/3713720/repo_lxml.py.zip)

In order to run that script I usually run something like:
```
export CONDA_DIR=<path to conda>
export PATH=$CONDA_DIR/bin:$PATH
python3 -mpip install pandas
python3 -mpip install openpyxl
python3 repo_lxml.py
```

mkurc-ant, 2019-10-11:
@litghost Thanks, It worked.

PR #1077 (2019-10-15): Pin libxml2 to version with fix.
Fixes #1066
closed 2019-10-15 by litghost