f4pga/f4pga-arch-defs #1068: Merge conflict prevents producing octopus merge

mithro, 2019-10-11:
If there is a conflict when trying to do an octopus merge then the merging WIP tool fails.

_Originally posted by @litghost in https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1067#issuecomment-540823673_

mithro, 2019-10-11:
https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/utils/update_tools/update_tools.py

litghost, 2019-10-16:
```
subprocess.CalledProcessError: Command 'cd /usr/local/google/home/keithrothman/cat_x/update_wip/vtr-verilog-to-routing && git merge wip/add-issue-template wip/avoid-criticality-issue wip/avoid_unordered_set wip/connection_box_serdes_merge_squashed wip/disable-check-route-option wip/equivalent-sites wip/genfasm-placeholders wip/prune_non_configurable_nodes wip/refactor_heap2 wip/rr-graph-binary-read-write wip/symbiflow-badger wip/use_expansion_for_place_delay && cd -' returned non-zero exit status 1.
```

New code doesn't handle failed octopus merge

closed 2019-10-23 by litghost