f4pga/f4pga-arch-defs #1071: Try SymbiFlow with full xc7a50t graph

mkurc-ant, 2019-10-11:


mkurc-ant, 2019-10-11:
First attempt:

- symbiflow-arch-defs @ 4d175925
- verilog-to-routing @  6d2166f2d
- https://github.com/SymbiFlow/libxml2-feedstock @ 7506dc8
- VPR patched with https://github.com/SymbiFlow/symbiflow-arch-defs/files/3705457/hack.patch.zip from https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1060 (manually cause `patch -p1 <hack.patch` failed)

failed with:

```
Traceback (most recent call last):
  File "/home/mkurc/Repos/google-symbiflow-arch-defs/xc7/utils/prjxray_routing_import.py", line 1095, in <module>
    main()
  File "/home/mkurc/Repos/google-symbiflow-arch-defs/xc7/utils/prjxray_routing_import.py", line 1085, in main
    import_graph_edges(conn, graph, node_mapping)
  File "/home/mkurc/Repos/google-symbiflow-arch-defs/utils/lib/rr_graph_xml/graph2.py", line 469, in serialize_edges
    serialize_edges(self.xf, edges)
  File "/home/mkurc/Repos/google-symbiflow-arch-defs/utils/lib/rr_graph_xml/graph2.py", line 115, in serialize_edges
    for src_node, sink_node, switch_id, metadata in edges:
  File "/home/mkurc/Repos/google-symbiflow-arch-defs/xc7/utils/prjxray_routing_import.py", line 766, in import_graph_edges
    (('fasm_features', check_feature(pip_name)), )
  File "/home/mkurc/Repos/google-symbiflow-arch-defs/xc7/utils/prjxray_routing_import.py", line 239, in check_feature
    return ' '.join((feature, find_hclk_cmt_hclk_feature(feature_path[0], m.group(1), m.group(2))))
  File "/home/mkurc/Repos/google-symbiflow-arch-defs/xc7/utils/prjxray_routing_import.py", line 190, in find_hclk_cmt_hclk_feature
    hclk_cmt_tile = HCLK_CMT_TILES[(hclk_tile, lr)]
KeyError: ('CLK_HROW_TOP_R_X60Y130', 'R')
xc7/archs/artix7/devices/CMakeFiles/file_xc7_archs_artix7_devices_rr_graph_xc7a50t_test.rr_graph.real.xml.dir/build.make:63: recipe for target 'xc7/archs/artix7/devices/rr_graph_xc7a50t_test.rr_graph.real.xml' failed
```

mkurc-ant, 2019-10-11:
Generation of `file_xc7_archs_artix7_devices_rr_graph_xc7a50t_test.rr_graph.real.xml` succeeded. Required **18-20GB** of RAM to complete.

mkurc-ant, 2019-10-12:
Generation of `*lookahead.bin` is taking more than 24h now...

litghost, 2019-10-12:
> Generation of `*lookahead.bin` is taking more than 24h now...

Oh dear :(  Can you see if you are hitting paging which is slowing everything down?

mkurc-ant, 2019-10-13:
Actually no, the memory usage stayed at ~8.5GB (+baseline on my system which was ~1.8GB). From the memory usage log I determnied that it took a little over **25h**

mkurc-ant, 2019-10-13:
And finally the VPR crashed on the placement stage:
```
## Build Device Grid took 0.06 seconds (max_rss 27.6 MiB, delta_rss +0.8 MiB)
## Loading routing resource graph
Starting build routing resource graph...
Warning 97: in check_rr_graph: fringe node 2436265 CHANY at (96,27) has no fanin.
         This is possible on a fringe node based on low Fc_out, N, and certain lengths.
## Loading routing resource graph took 40.27 seconds (max_rss 12184.5 MiB, delta_rss +12156.9 MiB)
  RR Graph Nodes: 35012terminate called after throwing an instance of 'kj::ExceptionImpl'
  what():  capnp/arena.c++:127: failed: Exceeded message traversal limit.  See capnp::ReaderOptions.
stack: 561ce8a71828 561ce8a72027 561ce8b1805e 561ce8a42c7c 561ce897c711 561ce8a60d1a 561ce893cde9 7fcde0e62b96 561ce89462b9
xc7/tests/buttons/CMakeFiles/buttons_basys3_x1y0_bin.dir/build.make:108: recipe for target 'xc7/tests/buttons/buttons_basys3_x1y0/artix7-xc7a50t-roi-virt-xc7a50t-test/top.place' failed
```

Seems that we've hit a limitation of the `capnp` lib...

duck2, 2019-10-13:
That's fixed easily by increasing the limit:
```c++
    /* Increase reader limit to 1G words. */
    ::capnp::ReaderOptions opts = ::capnp::ReaderOptions();
    opts.traversalLimitInWords = 1024 * 1024 * 1024;
    ::capnp::FlatArrayMessageReader reader(f.getData(), opts);
```

mkurc-ant, 2019-10-13:
@duck2 I made the patch and it worked!

With that patch I managed to successfully P&R the `buttons_basys3_x1y0` target without the ROI and graph limit! :fireworks: Though I'll be able to check whether it works on HW tomorrow.

Peak memory usage of the routing import script was **~20GB**. The VPR topped up at **~14GB**

mkurc-ant, 2019-10-14:
I've tried with murax, I've modified it to use only "R" and "M" IOBs but the routing failed with:
```
Warning 100: No routing path for connection to sink_rr 993270, retrying with full device bounding box
Cannot route from SYN-GND.GND[0] (RR node: 2 type: SOURCE location: (3,1) class: 0) to BLK-TL-SLICEM.D5[0] (RR node: 993270 type: SINK location: (98,76) class: 30) -- no possible path
Failed to route connection from '$false' to 'murax.system_uartCtrl.streamFifo_2.ram.4.0.0.dram1' for net 'murax.system_uartCtrl.uartCtrl_1.tx.io_configFrame_stop' (#0)
Routing failed.
```

mkurc-ant, 2019-10-14:
That's the problem...
```
# Set of (src, sink, switch_id) tuples that pip edges have been sent to
# VPR.  VPR cannot handle duplicate paths with the same switch id.
if use_roi:
    print('{} Adding synthetic edges'.format(now()))
    add_synthetic_edges(conn, graph, node_mapping, grid, synth_tiles)
```

mkurc-ant, 2019-10-14:
Building lookahead data again...

litghost, 2019-10-14:
Can you post the lookahead vpr_stdout log?  I'd like to see the relative breakdown between the router lookahead computation and the placer delay matrix computation.

mkurc-ant, 2019-10-14:
Unfortunately I overwrote it...

litghost, 2019-10-15:
I have data from running the lookahead on a 2/5 50T:
```
### Computing connection box lookahead map took 2715.01 seconds (max_rss 5840.1 MiB, delta_rss +1451.2 MiB)
### Computing delta delays took 3229.89 seconds (max_rss 5840.1 MiB, delta_rss +0.0 MiB)
```

I'm going to run a 1/5 50T and 4/5 50T now on the same VTR and same machine to see if we see the trend, and whether we are looking at a `O(N)` or `O(N^2)` behavior.

litghost, 2019-10-15:
@mkurc-ant I think https://github.com/litghost/symbiflow-arch-defs/commit/b0c9a52adb1c50a1f4a46a8f272d9aeff0c5ad2c should fix the two issues you've seen.  I'm testing with the full 50T as well to get more data points.

mkurc-ant, 2019-10-15:
@litghost Great. What strategy do you suggest for the `capnp` reader word limit? Should we make it a global constant, a parameter?

litghost, 2019-10-15:
> @litghost Great. What strategy do you suggest for the `capnp` reader word limit? Should we make it a global constant, a parameter?

@duck2's solution is fine.

mkurc-ant, 2019-10-15:
@litghost I have data for the full 50T when ran at my i7 @ 3.2GHz:

```
RR Graph Nodes: 3501251
RR Graph Edges: 22930679
```
```
### Computing connection box lookahead map took 8852.38 seconds (max_rss 20706.6 MiB, delta_rss +8521.7 MiB)
### Computing delta delays took 96715.86 seconds (max_rss 20706.6 MiB, delta_rss +0.0 MiB)
```

litghost, 2019-10-15:
Thanks.  So it looks like the connection box lookahead is scaling linearly (for the most part), as expected.  The placer delta delays is decidedly not. I'm going to add some instrumentation to try to understand what is going on there.  It is possible that the placer delta delays is still scaling linearly, but have a constant factor change.

litghost, 2019-10-15:
1/5 50T data:
```
## Computing placement delta delay look-up
  RR Graph Nodes: 757609
  RR Graph Edges: 4376487
### Computing connection box lookahead map took 1295.23 seconds (max_rss 3077.0 MiB, delta_rss +601.8 MiB)
### Computing delta delays took 1148.58 seconds (max_rss 3077.0 MiB, delta_rss +0.0 MiB)
```

litghost, 2019-10-15:
FYI, the titan bench mark does not take nearly this long to generate it's place delay matrix:
```
## Build routing resource graph took 574.85 seconds (max_rss 22892.0 MiB, delta_rss +493.2 MiB)
  RR Graph Nodes: 48967826
  RR Graph Edges: 390313690
## Computing router lookahead map
## Computing router lookahead map took 1595.44 seconds (max_rss 22892.0 MiB, delta_rss +0.0 MiB)
# Create Device took 2179.58 seconds (max_rss 22892.0 MiB, delta_rss +493.2 MiB)

# Placement
## Computing placement delta delay look-up
RR graph channel widths unchanged, skipping RR graph rebuild
### Computing delta delays
### Computing delta delays took 4202.33 seconds (max_rss 22892.0 MiB, delta_rss +0.0 MiB)
## Computing placement delta delay look-up took 4203.61 seconds (max_rss 22892.0 MiB, delta_rss +0.0 MiB)
```

It's definitely worth investigating what is going on with this routing graph that is significantly smaller in terms of nodes/edges, and yet takes a very long time to process.

For example, are there certain profiles that are taking a very long time, and could that be avoided, etc.

litghost, 2019-10-15:
As a quick sanity check, I ran `/usr/bin/time -v` on the graph build up to the output of `file_xc7_archs_artix7_devices_rr_graph_xc7a50t_test.rr_graph.virt.xml` and `file_xc7_archs_artix7_devices_rr_graph_xc7a50t_test.rr_graph.real.xml` to confirm some behaviors with the full 50T graph.

Wallclock time:
 - `file_xc7_archs_artix7_devices_rr_graph_xc7a50t_test.rr_graph.virt.xml` : 29:08.64
 - `file_xc7_archs_artix7_devices_rr_graph_xc7a50t_test.rr_graph.real.xml` : 38:49.56

Memory usage:
 - `file_xc7_archs_artix7_devices_rr_graph_xc7a50t_test.rr_graph.virt.xml` : 4.53 GiB
 - `file_xc7_archs_artix7_devices_rr_graph_xc7a50t_test.rr_graph.real.xml` : 18.66 GiB

This means that `prjxray_routing_import.py` is taking ~19 GiB all on it's own, which is should be fixed.

mkurc-ant, 2019-10-16:
I can confirm that. Same on my system.

closed 2019-12-15 by litghost