f4pga/f4pga-arch-defs #1091: Reduce excessive memory usage of the XRay routing import

mkurc-ant, 2019-10-18:
1. Determine why the `prjxray_routing_import.py` uses ~20GB of memory when working on the full rr 50T graph
2. Optimize it to use less.

mkurc-ant, 2019-10-18:
The `lib_rrgraph_xml` actually loads the whole XML (~950MB) to memory through the ElementTree (libxml) thus converting it into its internal representation. This is bad and very memory inefficient!

mkurc-ant, 2019-10-18:
I'm working on a solution. So far I managed to reduce the memory footprint for the whole 50T graph to only a few GB by iteratively parsing the XML instead of loading it into memory at once.

litghost, 2019-10-18:
> I'm working on a solution. So far I managed to reduce the memory footprint for the whole 50T graph to only a few GB by iteratively parsing the XML instead of loading it into memory at once.

Depending on how fast that solution ends up being, it may be worthwhile to think about using `xmllint` to filter the XML prior to even getting to the python.  It is unclear if `xmllint` can do iterative XML parsing + filtering.

mkurc-ant, 2019-10-21:
Here it is: https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1094

closed 2019-10-23 by litghost