f4pga/f4pga-arch-defs #1169: Setting bits for IOB tiles not used by a design

mkurc-ant, 2019-11-22:
During initial playing with IOSTANDARD different than LVCMOS33 in https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1162 I discovered the following:

When selecting a differential IOSTANDARD for a pair of IOBs that conflicts with eg. LVCMOS33 Vivado sets/clears some bits in all iobs in a given bank. It does it regardless whether they are used by the design or not.

Right now we do not have a mechanism in SymbiFlow which would implement a similar behavior. This also poses an issue with comparing bitstreams generated using SymbiFlow and Vivado.

litghost, 2019-11-22:
This kind of change probably belongs in fasm2frames.  Can you write up a detailed example of the behavior?  I don't think I understand what you are proposing is happening here.

mkurc-ant, 2019-11-27:
I've done some minitesting today. I've found that the following features are enabled for unused IOB sites when given IOSTANDARD+DRIVE+SLEW is set on the used ones from the same bank. I haven't enforced pullup/pulldown:

```
LVCMOS12_LVCMOS15_LVCMOS18_SSTL135.STEPDOWN
LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL.SLEW.FAST
PULLTYPE.PULLDOWN
```
for

- LVCMOS12
- LVCMOS18
- SSTL135

I haven't played with setting pullup/pulldown for all unused IOBs in the design as the result is pretty obvious.

The question is whether we should set the eg. STEPDOWN feature for unused IOBs in SymbiFlow and how not doing it will affect the design operation in hardware.

litghost, 2019-11-27:
FYI, `SLEW.FAST` and `PULLTYPE.PULLDOWN` are both NULL features (e.g. no bits are set).  So only `LVCMOS12_LVCMOS15_LVCMOS18_SSTL135.STEPDOWN` needs to be propigated to the bank.  I believe the logic is something akin to:
 - If `STEPDOWN` is used anywhere in the bank, all IOB's need to have `STEPDOWN` set
 - If the IOSTANDARD of any IOB requires that `STEPDOWN` is not set, that is an error (e.g. incompatible IOSTANDARD).

litghost, 2019-11-27:
I believe `STEPDOWN` needs to be set for `LVCMOS15` too?

mkurc-ant, 2019-11-27:
Uh, I forgot to use LVCMOS15 in the minitest... I'll check that.

mkurc-ant, 2019-11-28:
I checked more IOSTANDARDs and the feature `LVCMOS12_LVCMOS15_LVCMOS18_SSTL135.STEPDOWN` is set for unused IOBs of the same bank when using one of:

- LVCMOS12
- LVCMOS15
- LCVMOS18
- MOBILE_DDR (?)
- SSTL135
- SSTL135_R
- SSTL15
- SSTL15_R
- SSTL18_II
- HSUL_12