f4pga/f4pga-arch-defs #1172: Make a Yosys plugin that does the inout port split

mkurc-ant, 2019-11-25:
Since BLIF file cannot handle inout ports, these are represented as a single net connected to both input and output port with the same name. This creates an issue when something drives that net as there are multiple drivers there.

Currently to solve this we use the following flow:
- Synthesize the design using Yosys but write it to JSON instead of BLIF.
- Process the JSON using a python script that splits inout ports to separate input and output ports.
- Load the JSON to Yosys again and write the final BLIF (this is essentially a format conversion).

The "as it should be done" solution would be to have a plugin for Yosys that does the inout port split. That would allow to get rid of the need for writing JSONs and processing them.

mithro, 2019-11-25:
We already have eBLIF -- could we just extend that?

mkurc-ant, 2019-11-26:
One possibility is to add let's say an `.inout` model to the BLIF file. But that's only one thing. We would then need to make VPR support inout ports which it does not right now. I can't tell how much work would that require.