f4pga/f4pga-arch-defs #1182: PLL and BUFG need to be placed in the same clock row region

acomodi, 2019-11-26:
There are several situations for which the clock is being routed from the PLL output to the BUFG tile that are placed in incompatible locations.

Incompatible location example:
- PLL placed in TOP half (e.g. X1Y1)
- BUFG placed in BOTTOM half (e.g. X0Y0).

In such a situation, nor the BUFG output can be driven to the PLL clk input neither PLL output to BUFG input.

This situation causes fasm2bels to fail, as well a non-working output bitstream produced with the symbiflow toolchain.

acomodi, 2019-11-26:
@litghost FYI

litghost, 2019-12-15:
I've proposed #1231 as a solution to this issue.

acomodi, 2020-01-08:
Closing. Fixed with https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1237

closed 2020-01-08 by acomodi