f4pga/f4pga-arch-defs #1188: Getting started build example

Tcub, 2019-11-28:
Hi,

I have just had a go at following the getting started instructions on the .io page. When running the make counter_arty_bin command it would fail and the traceback stated it could not find and import the numpy module while running prjxray_edge_library.py

I go it working by calling the pip install inside /build/env/conda/bin/ and installing numpy with it. I also noticed that pip is called to install numpy in one of the build files for the ice40 so I was wondering if maybe it was an oversight for xc7 devices. (It also could be a weird setup thing my end or classic not reading the instructions but I thought it worth mentioning)

acomodi, 2019-12-04:
@Tcub To solve the issue without having to manually install numpy within conda, the following make target can be used:

```
make all_conda
```

I believe this should be added to the README

Tcub, 2019-12-05:
Yep that worked. Many thanks for the help.

acomodi, 2019-12-05:
Thanks for filing the issue. Closing.

closed 2019-12-05 by acomodi