f4pga/f4pga-arch-defs #1189: Kokoro CI target "VtR + SymbiFlow Architecture Definitions" uses wrong VPR version

mkurc-ant, 2019-11-29:
I've observed some CI tests failing recently for a not obvious reason. I've also noticed that the Kokoro target "VtR + SymbiFlow Architecture Definitions (Presubmit)" uses the upstream VPR instead of the conda package pointed at by CMakeLists.txt file.

Is the use of upstream VPR intentional there?

litghost, 2019-12-03:
I believe I see the issue, it is using master instead of master+wip.

litghost, 2019-12-03:
@mithro

litghost, 2019-12-03:
Issue has been resolved!

closed 2019-12-03 by litghost

mithro, 2019-12-03:
@litghost Have you confirmed Kokoro is now running on the right branch?

litghost, 2019-12-03:
> @litghost Have you confirmed Kokoro is now running on the right branch?

Yes, https://source.cloud.google.com/results/invocations/154bb11d-4c63-413c-83de-7ab80fbaaeb2/details

mithro, 2019-12-03:
Awesome!