f4pga/f4pga-arch-defs #1196: Flaky xc7 vendor tests

litghost, 2019-12-03:
There are currently two xc7 vendor CI flakes:

1:
```
ERROR: [Place 30-176] Unroutable Placement! The following clock source instance is driving the following locked load instances. The clock source instance is placed too far away from its load instance to be routable. 
	CLK_HROW_TOP_R_X60Y78_BUFHCE_X1Y18_BUFHCE (BUFHCE.O) is locked to BUFHCE_X1Y18
	The loads are distributed to 1 user pblock constraints. In addition, there are 20 loads not in user pblock constraints.

	Displaying only the first 20 or fewer instances under each constraint as list of loads is too long
```

And 2:

```
--- /tmpfs/src/github/symbiflow-arch-defs-presubmit-xc7-vendor/build/xc7/tests/bram_test/bram_test_36/artix7-xc7a50t-basys3-roi-virt-xc7a50t-basys3-test/top.bit.fasm	2019-12-03 11:17:42.766868834 -0800
+++ /tmpfs/src/github/symbiflow-arch-defs-presubmit-xc7-vendor/build/xc7/tests/bram_test/bram_test_36/artix7-xc7a50t-basys3-roi-virt-xc7a50t-basys3-test/design_bram_test_36_vivado.bit.fasm	2019-12-03 11:24:40.031996830 -0800
@@ -4159,7 +4159,6 @@
 INT_L_X6Y114.ER1BEG2.LOGIC_OUTS_L13
 INT_L_X6Y114.FAN_ALT0.GFAN0
 INT_L_X6Y114.FAN_ALT4.GFAN0
-INT_L_X6Y114.FAN_ALT5.FAN_BOUNCE1
 INT_L_X6Y114.GFAN0.GND_WIRE
 INT_L_X6Y114.GFAN1.GND_WIRE
 INT_L_X6Y114.IMUX_L10.GFAN0
```

I know what the issue is with #2, but I haven't been able to replicate #1 locally yet.

acomodi, 2019-12-04:
For the first issue my guess would be that the BUFHCE output traverses a clock region to get to another one, producing an `invalid` route.
This could happen only in the case the clock signal is being routed through logic cells, instead of the dedicated clock network as, to get to another clock region through the dedicated network, the signal should be correctly routed through another BUFHCE.

I suspect this is a kind of issue related to https://github.com/SymbiFlow/symbiflow-arch-defs/issues/1153

litghost, 2020-07-14:
https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1573

closed 2020-07-14 by litghost