f4pga/f4pga-arch-defs #1198: Litex pending PRs

acomodi, 2019-12-04:
This issue collects all the pending PRs needed to build the full LiteX design.

### Necessary

These are the `necessary` PRs, critical for having a working HW implementation of the design:

- [ ] IOBUF inference using Yosys: https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1170
- [ ] XDC Plugin integration: https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1158
- [ ] Mini Litex test (minimal version of the full LiteX design: https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1181
- [ ] LOC placement constraints: https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1186
- [ ] SERDES/IDELAY support: https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1174
- [ ] HCLK_IOI support (for the IDELAYCTRL): https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1191

### Not necessary

These are `not necessary` PRs, they do not directly impact the LiteX design build outcome:

- [ ] Minitest for different IOSTANDARDS: https://github.com/SymbiFlow/prjxray/pull/1157

litghost, 2019-12-04:
Status's as of 12/4:
- IOBUF inference using Yosys: #1170
  - Needs to be rebase and updated now that Yosys fixes have been merged into Yosys master+wip
- XDC Plugin integration: #1158
  - Outstanding review comments
- Mini Litex test (minimal version of the full LiteX design: #1181
  - Pending behind #1158 
- LOC placement constraints: #1186
  - CI is red and pending review comments
- SERDES/IDELAY support: #1174
  - CI is red
- HCLK_IOI support (for the IDELAYCTRL): #1191
 - Outstanding review comments

litghost, 2019-12-09:
Status as of 12/9:
- Mini Litex test (minimal version of the full LiteX design: #1181
 - Working on hardware, but needs LOC constraints to prevent BUFG and PLL from being placed in invalid locations.
- HCLK_IOI support (for the IDELAYCTRL): #1191
 - PR was split.  New PR is required to finish IDELAYCTRL support, as diff_fasm tests were failing and are currently disabled.

acomodi, 2020-12-08:
This issue has been solved and LiteX is enabledand supported. Closing.

closed 2020-12-08 by acomodi