f4pga/f4pga-arch-defs #1207: IDELAYCTRL ignored by VPR

acomodi, 2019-12-06:
The current LiteX design instantiates the IDELAYCTRL with inputs only. This causes VPR to ignore the relative nets.

By driving the output signal (`RDY`) to logic, VPR fails to route due to some issues with the `rr_graph`, ending in an unroutable situation:

```
# Routing
  RR Graph Nodes: 1288326
  RR Graph Edges: 7938537
Confirming router algorithm: TIMING_DRIVEN.
---- ------ ------- ---- ------- ------- ------- ----------------- --------------- -------- ---------- ---------- ---------- ---------- --------
Iter   Time    pres  BBs    Heap  Re-Rtd  Re-Rtd Overused RR Nodes      Wirelength      CPD       sTNS       sWNS       hTNS       hWNS Est Succ
      (sec)     fac Updt    push    Nets   Conns                                       (ns)       (ns)       (ns)       (ns)       (ns)     Iter
---- ------ ------- ---- ------- ------- ------- ----------------- --------------- -------- ---------- ---------- ---------- ---------- --------
Warning 186: No routing path for connection to sink_rr 245847, retrying with full device bounding box
Warning 187: No routing path for connection to sink_rr 551334, retrying with full device bounding box
Cannot route from BLK-TL-HCLK_IOI3.HCLK_IOI_IDELAYCTRL_RDY[0] (RR node: 501707 type: SOURCE location: (2,133) class: 3 capacity: 1) to BLK-TL-SLICEL.A2[0] (RR node: 551334 type: SINK location: (12,153) class: 1 capacity: 1) -- no possible path
Failed to route connection from 'serdes_test.idelayctrl' to '$auto$alumacc.cc:485:replace_alu$51.slice[0].carry4_1st_full' for net 'serdes_test.RDY' (#76)
Routing failed.
# Routing took 0.36 seconds (max_rss 2351.8 MiB, delta_rss +0.0 MiB)
Circuit is unroutable with a channel width factor of 500.
VPR failed to implement circuit
```

the `rr_graph_walk` tool did not find a possible route between the SOURCE and SINK nodes

litghost, 2019-12-10:
I took a look at Vivado, and the path from `IDELAYCTRL.RDY` is relatively short.  It must be a graph import bug or something.

litghost, 2019-12-10:
@acomodi Have you create an upstream VTR issue on this problem yet?

acomodi, 2019-12-10:
Not yet, I can open it right now.

litghost, 2019-12-10:
> Not yet, I can open it right now.

To be clear, the issue is VPR not pruning BELs with inputs and no outputs.

litghost, 2019-12-10:
@acomodi : I just remembered:
```
  --sweep_dangling_blocks {on, off}
                    Controls whether dangling blocks are removed from the netlist (Default: on)
```

Try turning off `sweep_dangling_blocks`.  This will leave IDELAYCTRL in place, even in the absence of the RDY signal working

acomodi, 2019-12-10:
Right, I think I had already tried this, but I need to double-check.

acomodi, 2019-12-10:
```
  VCC          : 2
  SYN-VCC      : 2
  GND          : 1
  SYN-GND      : 1
  IOB33S       : 5
  BLK-TL-IOPAD_S : 5
  BLK-TL-HCLK_IOI3 : 1
  REG_FDSE_or_FDRE : 24
  BLK-TL-IOPAD : 10
  BLUT         : 1
  C5LUT        : 3
  B5LUT        : 1
  CEUSEDMUX    : 8
  BLK-TL-SLICEL : 6
  BLK-TL-PLLE2_ADV : 1
  SRUSEDMUX    : 8
  SLICE_FF     : 8
  CARRY4_VPR   : 6
  IDELAYE2     : 1
  CE_VCC       : 32
  IOB33_MODES  : 20
  FF_FDSE_or_FDRE : 8
  IDELAYCTRL   : 1
  SLICEL0      : 6
  SLICEM_MODES : 2
  OSERDESE2    : 1
  IOBUF_VPR    : 1
  C_DRAM       : 2
  COMMON_LUT_AND_F78MUX : 1
  FDRE         : 32
  BLK-TL-CLK_BUFG_BOT_R : 1
  PLLE2_ADV    : 1
  outpad       : 10
  inpad        : 11
  ILOGICE3     : 1
  OLOGICE3     : 1
  SR_GND       : 32
  SLICEM       : 2
  A5LUT        : 3
  B_DRAM       : 1
  BLK-TL-SLICEM : 2
  ALUT         : 3
  IOB33M       : 5
  BUFGCTRL1    : 1
  CLUT         : 2
  BUFGCTRL10   : 1
  BLK-TL-IOPAD_M : 5
  COMMON_SLICE : 8
  A_DRAM       : 2
  ISERDESE2    : 1
  lut          : 7
  BUFGCTRL0    : 1
  BUFGCTRL_VPR : 4
  IOB33        : 10
  BUFGCTRL11   : 1

        EMPTY: # blocks: 0, average # input + clock pins used: 0, average # output pins used: 0
        BLK-TL-SLICEL: # blocks: 6, average # input + clock pins used: 13.3333, average # output pins used: 8.5
        BLK-TL-SLICEM: # blocks: 2, average # input + clock pins used: 10, average # output pins used: 6.5
        BLK-TL-BRAM_L: # blocks: 0, average # input + clock pins used: 0, average # output pins used: 0
        BLK-TL-IOPAD: # blocks: 10, average # input + clock pins used: 4.1, average # output pins used: 1.3
        BLK-TL-IOPAD_M: # blocks: 5, average # input + clock pins used: 0.6, average # output pins used: 0.4
        BLK-TL-IOPAD_S: # blocks: 5, average # input + clock pins used: 0.4, average # output pins used: 0.6
        BLK-TL-CLK_BUFG_BOT_R: # blocks: 1, average # input + clock pins used: 32, average # output pins used: 4
        BLK-TL-CLK_BUFG_TOP_R: # blocks: 0, average # input + clock pins used: 0, average # output pins used: 0
        BLK-TL-PLLE2_ADV: # blocks: 1, average # input + clock pins used: 32, average # output pins used: 3
        BLK-TL-HCLK_IOI3: # blocks: 1, average # input + clock pins used: 1, average # output pins used: 0
        SYN-VCC: # blocks: 2, average # input + clock pins used: 0, average # output pins used: 0.5
        SYN-GND: # blocks: 1, average # input + clock pins used: 0, average # output pins used: 1
Absorbed logical nets 73 out of 164 nets, 91 nets not absorbed.
FPGA sized to 150 x 162 (xc7a50t-bottom-test)
Device Utilization: 0.00 (target 1.00)
        Block Utilization: 0.00 Type: BLK-TL-SLICEL
        Block Utilization: 0.00 Type: BLK-TL-SLICEM
        Block Utilization: 0.10 Type: BLK-TL-IOPAD
        Block Utilization: 0.10 Type: BLK-TL-IOPAD_M
        Block Utilization: 0.10 Type: BLK-TL-IOPAD_S
        Block Utilization: 1.00 Type: BLK-TL-CLK_BUFG_BOT_R
        Block Utilization: 0.00 Type: BLK-TL-CLK_BUFG_TOP_R
        Block Utilization: 0.50 Type: BLK-TL-PLLE2_ADV
        Block Utilization: 0.50 Type: BLK-TL-HCLK_IOI3
        Block Utilization: 2.00 Type: SYN-VCC
        Block Utilization: 1.00 Type: SYN-GND

# Packing took 0.51 seconds (max_rss 39.7 MiB, delta_rss +0.0 MiB)
Error 1:
Type: Other
File: /tmp/really-really-really-really-really-really-really-really-really-really-really-really-really-long-path/conda/conda-bld/vtr_1575656400987/work/vpr/src/pack/pack.cpp
Line: 180
Message: Failed to find device which satisifies resource requirements required: BLK-TL-SLICEL: 6, BLK-TL-SLICEM: 2, BLK-TL-IOPAD: 10, BLK-TL-IOPAD_M: 5, BLK-TL-IOPAD_S: 5, BLK-TL-CLK_BUFG_BOT_R: 1, BLK-TL-CLK_BUFG_TOP_R: 0, BLK-TL-PLLE2_ADV: 1, BLK-TL-HCLK_IOI3: 1, SYN-VCC: 2, SYN-GND: 1 (available BLK-TL-SLICEL: 2754, BLK-TL-SLICEM: 867, BLK-TL-IOPAD: 102, BLK-TL-IOPAD_M: 48, BLK-TL-IOPAD_S: 48, BLK-TL-CLK_BUFG_BOT_R: 1, BLK-TL-CLK_BUFG_TOP_R: 1, BLK-TL-PLLE2_ADV: 2, BLK-TL-HCLK_IOI3: 2, SYN-VCC: 1, SYN-GND: 1)
The entire flow of VPR took 0.79 seconds (max_rss 39.7 MiB)
```

@litghost  IDELAYCTRL is there, but VPR failed on VCC blocks. It seems there are too many of them packed, therefore VPR cannot find a suitable device

litghost, 2019-12-10:
Maybe we can change the sweep logic to only sweep if there are 0 inputs and 0 outputs?  That second VCC is likely totally disconnected?

acomodi, 2019-12-10:
I'll look into that

acomodi, 2019-12-10:
@litghost I believe is easily doable actually, the sweeping mechanism checks only if the block has no outputs. I'll change, that, make a wip branch if it works, and integrate it.

litghost, 2019-12-10:
> @litghost I believe is easily doable actually, the sweeping mechanism checks only if the block has no outputs. I'll change, that, make a wip branch if it works, and integrate it.

Before you do all that, make sure it fixes our immediate issue.

acomodi, 2019-12-10:
Sure

litghost, 2019-12-10:
I've identified a bug in the routing import that will prevent both IDELAYCTRL RST and RDY from being connected to the graph. I'm looking into a fix.

litghost, 2019-12-10:
> I've identified a bug in the routing import that will prevent both IDELAYCTRL RST and RDY from being connected to the graph. I'm looking into a fix.

I've isolated the issue to be an error in the aliveness check.

acomodi, 2020-12-08:
The IDELAYCTRL can now be correctly placed and routed, closing this issue.

Fixing PRs:
- https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/1484
- https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1698

closed 2020-12-08 by acomodi